COMPASS-DOE/sensor-data-pipeline #3: Synoptic Shiny App To Do

stephpenn1, 2022-06-28:
### Goals
- Read in real time synoptic data from all (5) sites
- Create basic graphs for people to identify sensor errors and troubleshoot in the field
- Create tables to identify sensor-specific issues

### To Do
- [X] Create basic layout we want
- [ ] Figure out how to read in data from COMPASS drive
- [ ] Make sure we have a function for each data type that can read in a single `.dat` file
- [ ] Read in data in a loop (using `lapply`), saving as a temp file that gets overwritten (we don't want to save a ton of files to our local in the process) for each data type (e.g. AquaTroll, TEROS, etc.)
- [ ] Save data frames in a list
- [ ] Wrap that into a reactive variable called `reactive_list`

stephpenn1, 2022-07-01:
@stephpenn1 will work on adding sapflow data following Peter's aquatroll example

peterregier, 2022-07-01:
[] @peterregier will create a faceted plot of water depth

closed 2025-04-14 by stephpenn1