COMPASS-DOE/sensor-data-pipeline #12: L1a data - thoughts and questions

bpbond, 2022-11-23:
For after Thanksgiving...

L1a data are 
* Long form
* Matched with their design_link
* Converted to final units
* Have out-of-bounds flags attached

That's all fine. But what about...

## Structure and organization

Are they still in their datalogger/table files at this stage? Yes I think?

Probably at this stage they are sorted into `year_month_<datalogger> ` folders?

Are duplicate data allowed? Yes? (Assuming that de-duping happens when we go to L1b, the primary ready-for-analysis product.)

## Incomplete data

Because in the next step (to L1b) we need to combine _across_ dataloggers and tables, and we can't guarantee when data become available (dataloggers might be downloaded at different dates), it seems like data need to accumulate at L1a, sorted in their folders, and be reprocessed into L1b until a folder is deemed 'done'. At that point it's marked as do_not_process or something.

@selinalcheng How do you guys deal with this issue?

@stephpenn1 Let's discuss next time we talk :)

selinalcheng, 2022-11-28:
wrt structure and organization of L1A files, yep, they should still be in their datalogger/table files. At this point, we have the files named "datalogger_table_month_year.csv" and the files are sorted by project (just because that made the most sense for us) -- see attached photo. We also try to remove all duplicate data at the L1A stage.

[image]

Let me make sure I understand your question about incomplete data...
Let's say it's only mid-November, so our "datalogger_table_11_2022.csv" file only has dates up to 11/15/2022; it's an incomplete  monthly dataset. But we want to join it to other data....so what do we do?

This is a great question, and we don't have a system for this yet. I think that's because we don't really imagine a scenario like that happening. I think our formal joining and derived variable creation steps will happen after we collect all the data for one year. 

I'm not sure I see a huge problem with joining incomplete data since I assume you would be joining by timestamp (or rounded timestamp) anyway? The only thing is that once you get the full monthly dataset, you would have to re-join the now-complete data and overwrite the old monthly joined dataset. 

Maybe you could even have a system where in the initial join,

```
if(max(date(timestamp)) < last_day_of_month) {
# append "incomplete" to filename
}
```

And later, when running the join function on a whole directory of monthly files, if any filename has "incomplete" in it, rerun the join.

...does that make sense? Is this too convoluted? Happy to discuss on a call also!

bpbond, 2022-11-28:
**Thanks so much** @selinalcheng — yes, something like that makes sense to me. One way or another, we need the ability to re-run when new data come in, even though right now it seems like it'll be an unusual situation (as you note). 

Maybe you and I and @stephpenn1 can all discuss this in January? I think—but not sure what timeline Steph is planning—that we'll have a full trial system built out by then, all the way from raw data to analysis-ready products.

bpbond, 2022-11-28:
P.S.

```r
if(max(date(timestamp)) < last_day_of_month) {
# append "incomplete" to filename
}
```

Brilliant! 👏

selinalcheng, 2022-11-29:
Discussing in January sounds good to me! Awesome to hear that you guys have a plan to build out the system by then. You've sure put in a lot of work on this.

bpbond, 2023-02-06:
I think we're good here.

closed 2023-02-06 by bpbond