COMPASS-DOE/sensor-data-pipeline #17: Rearchitecting Portage sensor workflow

bpbond, 2023-02-01:
- [ ] `L1a.qmd` now combines across entire site and joins with plot table
- [ ] Implement units conversion based on real table
- [ ] Implement OOB flags based on real table
- [ ] Implement a derived variable. Defined through code? CSV file?
- [ ] Get feedback from SP at this point
- [ ] Redo L1b
- [ ] Feedback from SP, SC, RR

stephpenn1, 2023-02-01:
Loop me in when useful re: `Implement a derived variable`

bpbond, 2023-02-02:
@stephpenn1 

>Implement a derived variable

I know one case, which is we that want to compute water table level based on one of the Troll600 variables (plot specific) + a site level met variable. Is that right? 

Oh sap flux is another. I think it needs sap velocity (plot) + at least PAR (site) + tree level information (separate dataset).

Do you have other specific examples please? I'm trying to think through how this will work code-wise. 

Thank you

stephpenn1, 2023-02-02:
I need to think about this a bit more, but these should be the only variables:

| Variable         | External Data Needed |
|    :----:           |               :----:               |
| Sapflow         | ClimaVue (PAR), sapwood area         |
| Water Level (200 - nonvented)  | ClimaVue (atm pressure); inventory sheet  |
| Water Level (600 - vented)  |  inventory sheet  |

bpbond, 2023-02-03:
Start with a `derived_table.csv`

new_research_name | research_names_needed
-- | --
sapflow_avg | sapflow,sapflow_deep

Data enters derived calculation and is filtered for `research_names_needed` entries:

Site | Plot | Tree | TIMESTAMP | research_name | value
-- | -- | -- | -- | -- | --
Portage | Upland | 001 | 2022-11-09T00:00:00Z | sapflow | 0.598
Portage | Upland | 001 | 2022-11-09T00:15:00Z | sapflow | 0.599
Portage | Upland | 001 | 2022-11-09T00:30:00Z | sapflow | 0.599
Portage | Upland | 001 | 2022-11-09T00:00:00Z | sapflow_deep | 0.581
Portage | Upland | 001 | 2022-11-09T00:15:00Z | sapflow_deep | 0.583
Portage | Upland | 001 | 2022-11-09T00:30:00Z | sapflow_deep | 0.583

This gets reshaped before passing to handler function:

Site | Plot | Tree | TIMESTAMP | sapflow | sapflow_deep
-- | -- | -- | -- | -- | --
Portage | Upland | 001 | 2022-11-09T00:00:00Z | 0.598 | 0.581
Portage | Upland | 001 | 2022-11-09T00:15:00Z | 0.599 | 0.583
Portage | Upland | 001 | 2022-11-09T00:30:00Z | 0.599 | 0.583

If handler (`DERIVED_sapflow_avg()`) is called guaranteed that data columns are available. It computes `value` and assigns `units`:

Site | Plot | Tree | TIMESTAMP | sapflow | sapflow_deep | value | units
-- | -- | -- | -- | -- | -- | -- | --
Portage | Upland | 001 | 2022-11-09T00:00:00Z | 0.598 | 0.581 | n.nn | uuu
Portage | Upland | 001 | 2022-11-09T00:15:00Z | 0.599 | 0.583 | n.nn | uuu
Portage | Upland | 001 | 2022-11-09T00:30:00Z | 0.599 | 0.583 | n.nn | uuu

Back in the caller, we assign  a new `research_name` (given in the csv) and get of unneeded computational columns (the variables that went into the derived variable). 

Site | Plot | Tree | TIMESTAMP | value | units | research_name
-- | -- | -- | -- | -- | -- | --
Portage | Upland | 001 | 2022-11-09T00:00:00Z | n.nn | uuu | sapflow_avg
Portage | Upland | 001 | 2022-11-09T00:15:00Z | n.nn | uuu | sapflow_avg
Portage | Upland | 001 | 2022-11-09T00:30:00Z | n.nn | uuu | sapflow_avg

What about new `design_link`? Where does it come from?

bpbond, 2023-02-09:
Derived variables implemented in d7c521223d8c69b7d76bf2b31fb8bd364096a178. Output looks like this:

[image]

bpbond, 2023-02-09:
@stephpenn1 I tagged you requesting feedback in the [google doc](https://docs.google.com/document/d/1DsreUaGZDGIsurH_FVRHwgS0r53h7GTL0zdEOERbPqE/edit) but otherwise I think everything here is done for meeting with SERC folks next week, closing.

closed 2023-02-09 by bpbond