COMPASS-DOE/sensor-data-pipeline #18: Derived variables

bpbond, 2023-02-03:
Question @selinalcheng 

For computing arbitrary derived variables (see Steph's list [here](https://github.com/COMPASS-DOE/data-workflows/issues/17#issuecomment-1414382522)) I have been [thinking](https://github.com/COMPASS-DOE/data-workflows/issues/17#issuecomment-1414724566) and don't see a way to do this via csv tables and general code, in a way akin to the plot table merge or unit conversion.

Instead, we simply need a dedicated piece of code to compute each derived variable `z = ƒ(x, y, ...)` where `z` is something like sap flux or water level.

Does this match your experience and how you guys approach this? Thanks.

bpbond, 2023-02-03:
In case that's not clear: seems like we need dedicated functions `compute_sap_flux()` or `compute_water_level_200nv()`.

selinalcheng, 2023-02-06:
Hm...yeah, good question. We honestly do not have generalized functions for creating derived variables. Our system in that regard is pretty basic. We either use dedicated functions, as you mention like `compute_sap_flux()`, or we just directly create a new column akin to doing something like `df$derived_var <- df$col1 / df$col2`

closed 2023-02-06 by bpbond

bpbond, 2023-02-09:
Done in d7c521223d8c69b7d76bf2b31fb8bd364096a178