COMPASS-DOE/sensor-data-pipeline #20: Next steps

bpbond, 2023-02-15:
To do list from meeting today with @roylrich @selinalcheng @stephpenn1 

- [ ] preserve T05A information from raw
- [ ] output file names should include rows/cols of data
- [ ] output files name could include a brief hash
- [ ] duplicate data
- [ ] what are error codes for instruments? They should become NA
- [ ] "sn" - look for this - should travel with data
- [ ] TIMESTAMP_raw
- [ ] L1a has the raw timestamps as well, and not 15-min averages
- [ ] L1b has rounded timestamps, and only 15 min averages
- [ ] metadata - generate a file-level file

selinalcheng, 2023-10-31:
Roy says: you only need "sn" to travel with the data for the ExoTables, where I belei

bpbond, 2023-10-31:
Thanks! Here's an ExoTable header:

```
"TOA5","Compass_PTR_W_311","CR1000X","28723","CR1000X.Std.05.01","CPU:COMPASS_v3.3.CR1X","53163","ExoTable"
"TIMESTAMP","RECORD","PB","Statname","Date","Time","Conductivity","FDOM_QSU","FDOM_RFU","NLF_conductivity","ODO_sat","ODO_local","ODO_MgL","Pressure_psia","Salinity_PPT","Specific_Conductivity_uScm","Wiper_Position_mv","pH","pH_mv","Temp_C","Depth_m","Battery_v","Cable_v","Wiper_Current_ma","sn","snn"
"TS","RN","","","mmddyy","hhmmss","uS/cm","QSU","RFU","uS/cm","%Sat","%SatLocal","mg/L","psia","ppt","uS/cm","mv","pH","mv","Celsius","meters","volts","volts","ma","",""
"","","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp","Smp"
```

So, is "28723" the serial number here? 🙏

bpbond, 2023-10-31:
Bzzzzt OH I SEE you mean the "sn" and "snn" columns. Never mind, got it!

selinalcheng, 2023-10-31:
Oh yep, sorry! Those are the column headers.

bpbond, 2023-11-07:
All done here except for the error code issue which is also mentioned in #49

closed 2023-11-07 by bpbond