COMPASS-DOE/sensor-data-pipeline #41: SP thoughts on L1a sample data

stephpenn1, 2023-09-07:
- did a test reading in and plotting some teros data, seems pretty straightforward! we just need to make sure in our schema that we aren't missing and critical information in the design_link/research_name (e.g. the depth for the teros is critical info) that make the Level 1 datasets useful
- `ECO-FDOM-PTR-W` should be `EXO-xxxx` :)
- it would be nice to have all design links have four parts so that you can split in code easily, but if this isn't practical, no biggie

I propose the following structure for the data:

|                             |    L1a         |      L1b     |       L2     |
| :-------------     |       :----:  |     :---:     |     :---:    |
|**Type** |OOB & units|OOB & units|OOB & units & QC|
|**Organization** |by site/plot|by sensor|by "group" |
|**Orientation**    |long            |long        |wide        | 
|**Split**               | by month  | by month|by month|

bpbond, 2023-09-08:
Thanks for taking a close look!

>we just need to make sure in our schema that we aren't missing and critical information in the design_link/research_name (e.g. the depth for the teros is critical info) that make the Level 1 datasets useful

Agreed. We can do that through the names, and/or add column(s) to the `vars_metadata.csv` file with extra information to be brought into the Level 1 data.

>`ECO-FDOM-PTR-W`

🤦  #42 

>it would be nice to have all design links have four parts so that you can split in code easily

Do you mean that design links could have a regular and predictable four-part structure? I agree! Right now their inconsistency is annoying. What four parts are you thinking of?

>I propose the following structure for the data

What's the difference between your L1a and L1b? Is it just a question of how the data are sorted? I don't quite get it.

stephpenn1, 2023-09-11:
Difference is basically splitting each sensor into a new file. Thoughts? The motivation would be to have an option that cuts down on file size when the user doesn't need everything from a site. 

Example file names:
**L1a:** `Portage_20230901-20230902_L1a.csv`
**L1b:** `Portage_20230901-20230902_sapflow_L1b.csv`
**L1b:** `Portage_20230901-20230902_teros_L1b.csv`

stephpenn1, 2023-09-11:
>The motivation would be to have an option that cuts down on file size when the user doesn't need everything from a site.

Here I think we can make this decision based on how big we estimate the monthly file sizes to be.

bpbond, 2023-09-11:
Ah, okay, thanks. That's easy to do, but I'm not sure it deserves to be a different data level (i.e., L1b versus L1a) because the data haven't changed _at all_, they've just been split up differently.

bpbond, 2023-09-11:
Right now a full month of data for Portage is ~30 MB.

stephpenn1, 2023-10-01:
Okay that's not bad! I wonder what TEMPEST's would be like (I'd imagine we would split by plot?) which would decrease file size.

bpbond, 2023-10-01:
👍  Also note that the current L1a has columns we don't strictly need (e.g. `Units`, which is very repetitive and could be moved to a separate metadata file). Removing those drops the file size by over 50%, so <20 MB for a full month of Portage data.

I don't have a good feeling for how big TEMPEST will be. There are a LOT more sensors, right?

bpbond, 2023-10-16:
@stephpenn1 See email and [google doc table](https://docs.google.com/document/d/1zZjNWT19hnfD3IO2BmwXayrfO9ohji6_s3tmYVxsW04/edit).

bpbond, 2023-10-31:
I think we're good here? Thank you @stephpenn1 !

closed 2023-10-31 by bpbond

stephpenn1, 2023-10-31:
Yes we're good!