COMPASS-DOE/sensor-data-pipeline #43: More SP thoughts

stephpenn1, 2023-10-02:
@bpbond Below are my thoughts after running the driver script, I'd also like to suggest a few minor non-code-breaking changes if ok to do in a diff branch with a PR! THANK YOU for all of this - it's really robust 🎉

_Code thoughts_
- I know we will be, but a reminder that we have to be explicit on the version dependencies on the HPC (I had an issue with a new argument in `left_join`)
- My `left_join` error (happened at L1_normalize) printed in the console and not the driver log, also since it errored out there was not an L1_normalize log. It could be nice to capture any error and 
(1) notify in the driver log if an error was found (I think this is similar to what you have in the step files but not sure since it didn't work for my error) and 
(2) print the actual error message in the respective step’s log. I’m also curious what the current setup is for errors/logging and maybe my error just slipped through the cracks.

_For our meeting on the 11th_
- We should go through the units_bounds list with the group or have them look over our variable decisions before we finalize L1 for production
- I’m curious whether data users care if the data is split by sensor or plot

bpbond, 2023-10-02:
Thank you for the feedback @stephpenn1 ! Of course, feel free to open a PR or just push to `portage`.

_Code thoughts_

* AGREED we need to be careful about version dependencies, particularly if we are going to use `dplyr` which has a history of changing. One solution would be [renv](https://rstudio.github.io/renv/index.html); other would be to have the code check for minimal version numbers. 
* I will go back and take a look at capturing error messages and do some testing--thank you.
* Also note I added a response about data size [here](https://github.com/COMPASS-DOE/data-workflows/issues/41#issuecomment-1742094099)

_Meeting_

* I agree with your points above.
* More generally I guess, I have two questions: **what is our goal for the meeting** (what do we want to come out of it with), and **do we want to ask folks to come with anything** (for example, a list of the things they look for in QA/QC). Are we demo'ing the pipeline? Focusing on the L1a data? It would be great for us to have these clearly defined.

stephpenn1, 2023-10-04:
**Goal for the meeting next week:**
- What kinds of QC steps to do you do to your data when processing? _What other data do we need to do effective QC (L2) data on the data type? Is this user-ready data?_ 
- What kinds of "red flags" do you look for when monitoring sensors?

**Do we want to ask people to come with anything?**
- Take a look at L1? _Is this something you would use if L2 was available?_
- Take a look at units/bounds?
- We're going to ask about the QC steps you take to process your data: from sensor monitoring to analysis-ready data
-

stephpenn1, 2023-10-04:
For our meeting with RR and SC:
- How will we handle external data?
- How will we handle QC that is traditionally done visually?

bpbond, 2023-10-05:
@stephpenn1 Did we decide who would prepare slides? I'd suggest that I'll draft slides but you run meeting, but am flexible. Thank you

stephpenn1, 2023-10-05:
Could you make a google slide and share w me? I think its best if you add the info about the workflow and exampledata and ill form the discussion questions on there. And happy to present!

bpbond, 2023-10-16:
I think can close this.

closed 2023-10-16 by bpbond