COMPASS-DOE/sensor-data-pipeline #49: L1 beta checklist

bpbond, 2023-11-02:
Synthesized from the [Google doc](https://docs.google.com/document/d/1a4X0iuVeJUC1z-NWjlKSjMKxrAzloHmhS_BUSex80LA/edit?usp=sharing)

**From from SP/RR/SC step**

- [ ] Use site codes from [sample planning document](https://docs.google.com/spreadsheets/d/1-HLIjkx17gid0L81dvBoCiUqoqSDnUgoTA8m0w-zzPI/edit#gid=152374833), not names
- [ ] I removed units (available in metadata) and old_value (available in L0); remove `Site` column too? @stephpenn1  It's available in filename
- [ ] Improve template files so they look 'real'

**Feedback from PR and others step**

- [ ] Finish instrument bounds and units in [schema template](https://docs.google.com/spreadsheets/d/1ikOlTPCrrLe46Sn9ICS2hJ_5uV6Aav9k2kx_ChshHoQ/edit#gid=2056809878)
- [ ] Flag cleanup (see SP's notes in google doc)

bpbond, 2023-11-02:
Current file format:

[image]

stephpenn1, 2023-11-06:
> I removed units (available in metadata) and old_value (available in L0); remove Site column too? @stephpenn1 It's available in filename...but maybe this would be a pain for users

- Units: I think we should make the unit format a question on the beta google form (i.e should we put them in the variable name, keep as separate column, or put into metadata file)
- Site: it's in the design link too, so I'm more inclined to remove this column

stephpenn1, 2023-11-06:
not sure if we've added this in, but an additional L1 flag could be the -99999 reading on certain sensors (see meeting [doc](https://docs.google.com/document/d/1rIgVzDFjGVDUKIytWGx_l7pbgsuMGnNrDPKZOsTzIBI/edit?usp=sharing))

bpbond, 2023-11-10:
- [ ] Metadata files need to specify what the NA value is

bpbond, 2023-12-22:
We're good here

closed 2023-12-22 by bpbond