COMPASS-DOE/sensor-data-pipeline #52: TEROS QA/QC steps suggestions for L1

robertapeixoto, 2023-11-09:
Roberta -> These are the steps I'm currently looking for in the TEROS manual QA/QC process. They can help create an automated pipeline for simple flag L1.

- [ ]  Identifying SWC values below 0.1. They can suggest problems in the sensor or sensors out of the soil. 

- [ ] Identifying outliers based on the mean absolute deviation. These don't necessarily need to be filtered out at L2 (e.g. CB sites with high EC!), but they should be flagged for closer examination, especially LE.

- [ ] Issues reported by the technicians during their  [weekly checks](https://docs.google.com/spreadsheets/d/1An0kScGPSsWdejV2y93kpbXxoxXY5k_Z2orcKCL_iJY/edit?usp=drive_link). I've come across instances of 'low connectivity' and a month of missing data. Typically, there's no recorded data or no apparent problems. We might want to consider establishing another system or metadata to link these notes to the automated data processing pipeline. Maybe a power or connectivity check can anticipate problems here.

bpbond, 2024-04-02:
I have changed the issue title to refect that these are L2 items, not L1 (except perhaps the first). Thanks again @robertapeixoto !

bpbond, 2025-06-24:
I think this has all been subsumed by the v2-0 development work. Thank you @robertapeixoto !

closed 2025-06-24 by bpbond