COMPASS-DOE/sensor-data-pipeline #66: NWRC?

bpbond, 2023-12-04:
@roylrich What are these files in `COMPASS_PNNL_Rawdata_Archive`? Is the synoptic pipeline responsible for processing them?
```
NWRC_4757_TableDT_20220824143838.dat
NWRC_4757_TableDT_20220910000027.dat
NWRC_4757_TableDT_20221010000036.dat
NWRC_4757_TableDT_20221109000348.dat
NWRC_4757_TableDT_20221209000347.dat
NWRC_4757_TableDT_20230108000332.dat
NWRC_4757_TableDT_20230207000321.dat
NWRC_4757_TableDT_20230309000323.dat
NWRC_4757_TableDT_20230408000435.dat
NWRC_4757_TableDT_20230508000520.dat
NWRC_4757_TableDT_20230607000511.dat
NWRC_4757_TableDT_20230707000438.dat
NWRC_4757_TableDT_20230806000521.dat
NWRC_4757_TableDT_20230905000508.dat
NWRC_4757_TableDT_20231005000524.dat
NWRC_4757_TableDT_20231104000710.dat
NWRC_4757_TableDT_20231204000713.dat
NWRC_4757_TableDY_20220824143839.dat
NWRC_4757_TableDY_20220910000027.dat
NWRC_4757_TableDY_20221010000036.dat
NWRC_4757_TableDY_20221109000351.dat
NWRC_4757_TableDY_20221209000350.dat
NWRC_4757_TableDY_20230108000335.dat
NWRC_4757_TableDY_20230207000323.dat
NWRC_4757_TableDY_20230309000326.dat
NWRC_4757_TableDY_20230408000438.dat
NWRC_4757_TableDY_20230508000523.dat
NWRC_4757_TableDY_20230607000513.dat
NWRC_4757_TableDY_20230707000441.dat
NWRC_4757_TableDY_20230806000524.dat
NWRC_4757_TableDY_20230905000511.dat
NWRC_4757_TableDY_20231005000527.dat
NWRC_4757_TableDY_20231104000714.dat
NWRC_4757_TableDY_20231204000718.dat
NWRC_4757_TableHR_20220824143839.dat
NWRC_4757_TableHR_20220910000027.dat
NWRC_4757_TableHR_20221010000037.dat
NWRC_4757_TableHR_20221109000354.dat
NWRC_4757_TableHR_20221209000353.dat
NWRC_4757_TableHR_20230108000337.dat
NWRC_4757_TableHR_20230207000326.dat
NWRC_4757_TableHR_20230309000328.dat
NWRC_4757_TableHR_20230408000440.dat
NWRC_4757_TableHR_20230508000526.dat
NWRC_4757_TableHR_20230607000516.dat
NWRC_4757_TableHR_20230707000443.dat
NWRC_4757_TableHR_20230806000527.dat
NWRC_4757_TableHR_20230905000514.dat
NWRC_4757_TableHR_20231005000530.dat
NWRC_4757_TableHR_20231104000716.dat
NWRC_4757_TableHR_20231204000722.dat
NWRC_4757_TableTC_20220824143839.dat
NWRC_4757_TableTC_20220910000028.dat
NWRC_4757_TableTC_20221010000037.dat
NWRC_4757_TableTC_20221109000357.dat
NWRC_4757_TableTC_20221209000356.dat
NWRC_4757_TableTC_20230108000340.dat
NWRC_4757_TableTC_20230207000328.dat
NWRC_4757_TableTC_20230309000331.dat
NWRC_4757_TableTC_20230408000443.dat
NWRC_4757_TableTC_20230508000529.dat
NWRC_4757_TableTC_20230607000519.dat
NWRC_4757_TableTC_20230707000446.dat
NWRC_4757_TableTC_20230806000530.dat
NWRC_4757_TableTC_20230905000517.dat
NWRC_4757_TableTC_20231005000532.dat
NWRC_4757_TableTC_20231104000719.dat
NWRC_4757_TableTC_20231204000726.dat
NWRC_4757_TableTDP_20220824143839.dat
NWRC_4757_TableTDP_20220910000028.dat
NWRC_4757_TableTDP_20221010000037.dat
NWRC_4757_TableTDP_20221109000359.dat
NWRC_4757_TableTDP_20221209000359.dat
NWRC_4757_TableTDP_20230108000343.dat
NWRC_4757_TableTDP_20230207000330.dat
NWRC_4757_TableTDP_20230309000333.dat
NWRC_4757_TableTDP_20230408000445.dat
NWRC_4757_TableTDP_20230508000532.dat
NWRC_4757_TableTDP_20230607000521.dat
NWRC_4757_TableTDP_20230707000448.dat
NWRC_4757_TableTDP_20230806000533.dat
NWRC_4757_TableTDP_20230905000519.dat
NWRC_4757_TableTDP_20231005000535.dat
NWRC_4757_TableTDP_20231104000722.dat
NWRC_4757_TableTDP_20231204000730.dat
NWRC_TRAN_TableDT_20230408000448.dat
NWRC_TRAN_TableDT_20230607000524.dat
NWRC_TRAN_TableDT_20230707000451.dat
NWRC_TRAN_TableDT_20230806000536.dat
NWRC_TRAN_TableDT_20230905000522.dat
NWRC_TRAN_TableDT_20231005000538.dat
NWRC_TRAN_TableDT_20231104000725.dat
NWRC_TRAN_TableDT_20231204000734.dat
NWRC_TRAN_TableDY_20230408000450.dat
NWRC_TRAN_TableDY_20230607000527.dat
NWRC_TRAN_TableDY_20230707000453.dat
NWRC_TRAN_TableDY_20230806000539.dat
NWRC_TRAN_TableDY_20230905000525.dat
NWRC_TRAN_TableDY_20231005000541.dat
NWRC_TRAN_TableDY_20231104000728.dat
NWRC_TRAN_TableDY_20231204000738.dat
NWRC_TRAN_TableHR_20230408000453.dat
NWRC_TRAN_TableHR_20230607000529.dat
NWRC_TRAN_TableHR_20230707000456.dat
NWRC_TRAN_TableHR_20230806000542.dat
NWRC_TRAN_TableHR_20230905000527.dat
NWRC_TRAN_TableHR_20231005000544.dat
NWRC_TRAN_TableHR_20231104000731.dat
NWRC_TRAN_TableHR_20231204000742.dat
NWRC_TRAN_TableTC_20230408000455.dat
NWRC_TRAN_TableTC_20230607000532.dat
NWRC_TRAN_TableTC_20230707000458.dat
NWRC_TRAN_TableTC_20230806000545.dat
NWRC_TRAN_TableTC_20230905000530.dat
NWRC_TRAN_TableTC_20231005000546.dat
NWRC_TRAN_TableTC_20231104000734.dat
NWRC_TRAN_TableTC_20231204000746.dat
NWRC_TRAN_TableTDP_20230408000458.dat
NWRC_TRAN_TableTDP_20230607000535.dat
NWRC_TRAN_TableTDP_20230707000501.dat
NWRC_TRAN_TableTDP_20230806000548.dat
NWRC_TRAN_TableTDP_20230905000533.dat
NWRC_TRAN_TableTDP_20231005000549.dat
NWRC_TRAN_TableTDP_20231104000737.dat
NWRC_TRAN_TableTDP_20231204000750.dat
NWRC_WAX_TableDT_20230408000501.dat
NWRC_WAX_TableDT_20230508000535.dat
NWRC_WAX_TableDT_20230607000537.dat
NWRC_WAX_TableDT_20230707000503.dat
NWRC_WAX_TableDT_20230806000551.dat
NWRC_WAX_TableDT_20230905000536.dat
NWRC_WAX_TableDT_20231005000552.dat
NWRC_WAX_TableDT_20231104000740.dat
NWRC_WAX_TableDT_20231204000754.dat
NWRC_WAX_TableDY_20230408000503.dat
NWRC_WAX_TableDY_20230508000538.dat
NWRC_WAX_TableDY_20230607000540.dat
NWRC_WAX_TableDY_20230707000506.dat
NWRC_WAX_TableDY_20230806000554.dat
NWRC_WAX_TableDY_20230905000538.dat
NWRC_WAX_TableDY_20231005000555.dat
NWRC_WAX_TableDY_20231104000743.dat
NWRC_WAX_TableDY_20231204000758.dat
NWRC_WAX_TableHR_20230408000506.dat
NWRC_WAX_TableHR_20230508000541.dat
NWRC_WAX_TableHR_20230607000543.dat
NWRC_WAX_TableHR_20230707000508.dat
NWRC_WAX_TableHR_20230806000557.dat
NWRC_WAX_TableHR_20230905000541.dat
NWRC_WAX_TableHR_20231005000558.dat
NWRC_WAX_TableHR_20231104000746.dat
NWRC_WAX_TableHR_20231204000802.dat
NWRC_WAX_TableTC_20230408000508.dat
NWRC_WAX_TableTC_20230508000544.dat
NWRC_WAX_TableTC_20230607000545.dat
NWRC_WAX_TableTC_20230707000510.dat
NWRC_WAX_TableTC_20230806000559.dat
NWRC_WAX_TableTC_20230905000544.dat
NWRC_WAX_TableTC_20231005000601.dat
NWRC_WAX_TableTC_20231104000749.dat
NWRC_WAX_TableTC_20231204000806.dat
NWRC_WAX_TableTDP_20230408000511.dat
NWRC_WAX_TableTDP_20230508000547.dat
NWRC_WAX_TableTDP_20230607000548.dat
NWRC_WAX_TableTDP_20230707000513.dat
NWRC_WAX_TableTDP_20230806000603.dat
NWRC_WAX_TableTDP_20230905000547.dat
NWRC_WAX_TableTDP_20231005000603.dat
NWRC_WAX_TableTDP_20231104000752.dat
NWRC_WAX_TableTDP_20231204000810.dat
```

roylrich, 2023-12-04:
Ben,
I have a batch script that ends current data on a set day each month, renames it and puts it in the rawdataarchive. This keeps current data from getting too big. The batching takes whatever is in the current file so it could be more data than a single month on some occasions. It is named by tablename,  date archived,  and number of records included.   The NWRC files are data from some sapflow data that belongs to other MSM folks (glen and ken's group). I put them on network for their convenience. We are not responsible for processing.
Roy

External Email - Exercise Caution

bpbond, 2023-12-04:
Awesome; thank you.

closed 2023-12-04 by bpbond