COMPASS-DOE/sensor-data-pipeline #70: L1 bounds checking

bpbond, 2023-12-08:
Prompted by questions from @peterregier and @stephpenn1 -

It would make more sense to do the checking on original units, seems like. I mean, it works either way, but since we’re defining this as instrument​ limits, it’s more intuitive to do it in the instrument's native units…?

stephpenn1, 2023-12-08:
I tend to agree, if we ask for calibration ranges from the manual, we should flag based on that same unit.

For example, sapflow is a super simple flag - all values regardless of ecosystem should be between 0-1 mV (raw units). But once you transform to final units, this range varies by species, ecosystem, etc. since the unit transformation is dependent on each signal/sensor for each day. We'd then have to create some sort of algorithm to determine absolute ranges in g/m2/s, not ideal and no longer objective imo.

I imagine this can be a simple fix if we keep the original unit values that way we can flag based on the OG units and new units as needed, then remove the OG unit column.

bpbond, 2023-12-08:
I realized there is one good reason to use the converted units: the bounds appear in the metadata file and that makes it very easy for readers to interpret.

stephpenn1, 2023-12-21:
I gotcha - although I still feel it's more defensible to say we flagged OOB based on raw values via the manual ranges. I'll chat with Peter and Fausto about this today and see if any of their data types have a similar issue to sapflow

bpbond, 2024-01-19:
I think we're good here, at least for now?

closed 2024-01-19 by bpbond

stephpenn1, 2024-01-19:
Yup good for now