COMPASS-DOE/sensor-data-pipeline #71: Looking for TEMPEST design info

bpbond, 2023-12-09:
Hi @roylrich 

I am looking for the information I'll need for the TEMPEST design table--wiring/design info by datalogger, along with what plot each datalogger is in. Searching the google drive isn't turning up much...can you point me to this? 

Thanks!

roylrich, 2023-12-09:
@stephpenn1  and stepahie should be included in this. I did not due the original wiring but rebuilt on top of sapflow system.  Data loggers are ordered by site: Teros checks list should have those sensor information , and same for sapflow.  
Loggers  11, 12, 13 are the same as pakbus (pb) 41, 42, and 43 and are in control
Loggers  21, 22, 23, are the same as pakbus (pb) 44, 45, and 46 and are fresh
Loggers 31, 32, 33 are the same as pakbus(pb) 47,4 8, and 49 and are salt. 
I have asked stephanie wilson to look for this on SERC drives or google files not shared with me .

roylrich, 2023-12-09:
*stephanie wilson

bpbond, 2023-12-09:
That is @wilsonsj100 on GitHub 😄 

OK, that's great to have the plot assignments, thank you @roylrich . I will follow up with SP and SW and @aestearns01 about the Teros and sapflow mappings.

wilsonsj100, 2023-12-11:
Okay I see what we are looking for above. I will reach out to Alice/Evan and see if there is a document, but if not we can have them go out and grab this info.

bpbond, 2023-12-11:
Hi @wilsonsj100 -

Thank you! Just to make sure it's clear: I can see from the datalogger files that the `PNNL_11` datalogger has 14 "DiffVol_Avg" outputs for sapflow, but I don't know is what trees and depths those 1-14 channels are connected to. Similarly, I can see 22 three-channel TEROS outputs but don't know what specific probes they're connected to. Et cetera.

Thank you!

bpbond, 2023-12-11:
@roylrich 

>Loggers 11, 12, 13 are the same as pakbus (pb) 41, 42, and 43 and are in control
>Loggers 21, 22, 23, are the same as pakbus (pb) 44, 45, and 46 and are fresh
>Loggers 31, 32, 33 are the same as pakbus(pb) 47,4 8, and 49 and are salt.

What about `PNNL_41`? It appears to have sapflow, TEROS, and Aquatroll connected to it.

Also, what about these two loggers:
```
CR1000_2	Table1	Format
CR1000_2	Table1	RECORD
CR1000_2	Table1	Statname
CR1000_2	Table1	BattV
CR1000_2	Table1	RedoxRa({1:20})
CR1000_2	Table1	RedoxRb({1:20})
		
RedoxMU3_83	Table1	Format
RedoxMU3_83	Table1	RECORD
RedoxMU3_83	Table1	Statname
RedoxMU3_83	Table1	BattV
RedoxMU3_83	Table1	RedoxRa({1:20})
RedoxMU3_83	Table1	RedoxRb({1:20})
```

aestearns01, 2023-12-12:
I'll email you all the link to the TEMPEST Sap Flow Monitoring Doc. This
should have all relevant information about that network.  As far as maps,
are you looking for existing maps OR info to create maps? And are you
looking for lat-longs or just grid locations? I couldn't find the
initial survey data from 2020 (before my time) but I do have some survey
data that I did back in winter 2021/2022 and have lat/long for grid poles
in FW and SW and TEROS in FW.

bpbond, 2023-12-12:
Thank you! For now, I don't care about physical locations; but I need to know is what tree/depth each datalogger channel is measuring. Does that help?

aestearns01, 2023-12-12:
Yes, it does.

aestearns01, 2023-12-12:
@bpbond PNNL_41 was the logger designation for the TEMPEST Shoreline. It is now Logger 432 (pakbus 50) and is part of the Synoptic sensor network. That change occurred in Feb ?? 2023 . It does have SF, Teros and AQ associated with it. That data is backed up with the Synoptic Logger files on the G:drive.

bpbond, 2023-12-12:
Thank you @aestearns01 . What's the difference between "Logger" and "pakbus" -- what does the latter mean exactly?

closed 2023-12-13 by bpbond

aestearns01, 2023-12-13:
@bpbond the Logger number/ID is for the actual data logger (CR1000; CR1000X; etc) in each logger box that the sensors are directly wired into giving us the port/channel numbers associated with each sensor. The Pakbus ID is a data communication ID that is associated with each data logger. I'm not sure if you need the pakbus information for your purposes but check with @roylrich and team.

bpbond, 2023-12-13:
@aestearns01 Thank you! I thought I could ignore the pakbus but good to have that confirmed.