COMPASS-DOE/sensor-data-pipeline #73: Initial GCW datalogger table oddity

bpbond, 2023-12-11:
The beginning of the `GCW` datalogger outputs is:

```
COMPASS_GCW_W_431_TerosTableA_20221109000108.dat
COMPASS_GCW_W_431_TerosTableB_20221109000111.dat
COMPASS_GCW_W_431_TerosTableC_20221109000115.dat
COMPASS_GCW_W_431_TerosTableC_20221209000123.dat
COMPASS_GCW_W_431_TerosTableC_20230108000111.dat
COMPASS_GCW_W_431_TerosTableC_20230207000111.dat
COMPASS_GCW_W_431_TerosTableC_20230309000116.dat
COMPASS_GCW_W_431_TerosTableC_20230408000119.dat
```

@roylrich @wilsonsj100 Do you know what's up with those initial "A" and "B" tables, and then it becomes "C" for the result of time? Can we simply ignore/delete them?

wilsonsj100, 2023-12-11:
I am not sure - that might be a @roylrich question....

roylrich, 2023-12-11:
So these are the same data tables just signify a different SDI12 address for aquatrolls, it is just that we have the ability to add more modules to each logger if possible (say we wanted to put more wells in). I can manually turn these tables on with flags but they automatically turn each on based on which pakbus they are.  Does this make sense. If data exists, it can be combined among a,b,c for now for each  logger for now

roylrich, 2023-12-11:
You can check if data is NAN, which is likely on those tables

roylrich, 2023-12-11:
Only one of these tables has any real data

bpbond, 2023-12-11:
I don't totally understand but yes, the A and B tables just have NANs. So I'm going to ignore them. Thank you!

closed 2023-12-11 by bpbond