COMPASS-DOE/sensor-data-pipeline #88: Where are these non-2023 SWH data coming from?!?

bpbond, 2023-12-21:
The following folders of data are being written by L1:
```
SWH_2024
SWH_2032
SWH_2023
SWH_2019
```

bpbond, 2023-12-21:
This looks like a problem in the raw data?

[image]

@roylrich Is this a known thing?

roylrich, 2023-12-21:
I looked at this whole file. The loggers were placed for a while before the instrumentation attached. As soon as it has data, the time shifts to correct time in this file. The time is coordinated by server and so we did not have coms at transition until cell was installed later on.  The out time data could be removed in this case with no loss. 
RR

bpbond, 2023-12-21:
Thanks. Is there a known "good data start" point in time?

Also, is it OK if I do delete the bad-timestamp data right in the original Dropbox data? I don't want it creeping back in later.

roylrich, 2023-12-21:
Good start:
 "2023-04-03 15:00:00",975,422,"COMPASS_SWH_TR","813IN-SITU AT600V232 1",763.9,13.08,1858.9,2406.8,1.24,1.56,1,537.96,0,482.3,195.3,0.34,3.21,3.9,1.086,95.48,11.371,97

I actually need to think about the best way to do this. The data is mirrored to dropbox and google drive from a folder on my server. I think it would be replaced by the file from my server if we change it on dropbox.

bpbond, 2023-12-21:
Okay. I'm going to leave this issue open for now, then. Thanks.

roylrich, 2024-01-17:
Is this just in this file or can you give me a list of files with this date. I will be do edits on server when I get on next

bpbond, 2024-01-17:
@roylrich Most of the initial 2023-04-08 files have this problem:
* Compass_SWH_TR_422_CheckTable_20230408000350.dat - lines 5-976 are bad
* Compass_SWH_TR_422_SapflowB_20230408000353.dat - lines 5-976 are bad
* Compass_SWH_TR_422_TerosTableA_20230408000356.dat - file OK but no data
* Compass_SWH_TR_422_TerosTableB_20230408000358.dat - lines 5-976 are bad
* Compass_SWH_TR_422_WaterLevel600A_20230408000401.dat - file OK but no data
* Compass_SWH_TR_422_WaterLevel600C_20230408000406.dat - file OK but no data
* Compass_SWH_UP_423_CheckTable_20230408000409.dat  - lines 5-4029 are bad
* Compass_SWH_UP_423_SapflowA_20230408000412.dat  - lines 5-4029 are bad
* Compass_SWH_UP_423_TerosTableA_20230408000414.dat  - lines 5-4029 are bad
* Compass_SWH_UP_423_WaterLevel600A_20230408000417.dat  - lines 5-4029 are bad
* Compass_SWH_W_421_CheckTable_20230408000419.dat - lines 5-3744 are bad
* Compass_SWH_W_421_ClimaVue50_15min_20230408000422.dat - lines 5-3744 are bad
* Compass_SWH_W_421_ClimaVue50_24hr_20230408000425.dat  - lines 5-43 are bad
* Compass_SWH_W_421_ExoTable_20230408000427.dat  - lines 5-4029 are bad
* Compass_SWH_W_421_TerosTableC_20230408000430.dat  - lines 5-4029 are bad
* Compass_SWH_W_421_WaterLevel600C_20230408000432.dat  - lines 5-3744 are bad

bpbond, 2024-01-17:
It might be simplest to delete all the 20230408 files.

bpbond, 2024-01-19:
For the 0-9 release, I will simply drop the April 8th files from SWH. @roylrich will deal with this on his end by editing/deleting.

closed 2024-01-20 by bpbond