COMPASS-DOE/sensor-data-pipeline #97: What's the initial version number?

bpbond, 2024-01-17:
Are we announcing v1.0 of the sensor data next week? Or declaring the data still in beta for a couple months, so v0.9 or something?

stephpenn1, 2024-01-18:
I think since we anticipate not having everything filled out to 100%, we should start with 0.9. I will add to my to-do list to create a similar readme to the EXCHANGE one that includes a versioning table for people to reference in the google drive

bpbond, 2024-01-18:
I agree. Thank you @stephpenn1 

Re versioning table — right now I have a "changelog" in the README [here](https://github.com/COMPASS-DOE/data-workflows/blob/main/synoptic/data_TEST/L1_metadata/README_v%3F%3F%3F.txt), to be distributed with every data release. Does that work?

stephpenn1, 2024-01-18:
Ah yeah, does this get automatically produced?

bpbond, 2024-01-18:
Yes, automatically included with each release. We need to write the changelog entry itself, of course.

stephpenn1, 2024-01-18:
Okay yes we'll just stick with this then. Thanks!

bpbond, 2024-01-18:
Thanks so much!

closed 2024-01-18 by bpbond