COMPASS-DOE/sensor-data-pipeline #108: 1min and 5min datalogger files

bpbond, 2024-01-19:
@roylrich Could you remind me what the deal is the the "1min" and "5min" datalogger files? We are going to process them through the pipeline, right?

E.g. `PNNL_31_sapflow_1min_20210816000006.dat`

Thanks

roylrich, 2024-01-19:
> @roylrich Could you remind me what the deal is the the "1min" and "5min" datalogger files? We are going to process them through the pipeline, right?
> 
> E.g. `PNNL_31_sapflow_1min_20210816000006.dat`
> 
> Thanks

These were setup for events, I don't know if anyone uses them? Could we process for maybe 1 month around tempest evenst?

bpbond, 2024-01-19:
If they're valid data, all good, let's process! No reason not to.

closed 2024-01-19 by bpbond