COMPASS-DOE/sensor-data-pipeline #119: TEMPEST TEROS EC data values look wrong

peterregier, 2024-03-25:
L1 TEROS data in 2023-TMP are order of magnitude of 10^6 for electrical conductivity. I would expect ranges of 100-10000 for uS/cm or <<1 to ~10 for mS/cm.

bpbond, 2024-04-01:
Thanks @peterregier ! The corresponding entry in our [metadata file](https://github.com/COMPASS-DOE/data-workflows/blob/main/synoptic/metadata/L1_metadata/L1_metadata_vars.csv) shows:

<!DOCTYPE html>
TerosTable | EC | soil_EC_5cm | dS/m | uS/cm | x * 100000 | 0 | 20000 | Soil electrical conductivity at 5 cm
-- | -- | -- | -- | -- | -- | -- | -- | --

which means we're multiplying the raw values by 100,000 to arrive at final units of µS/cm, with an expected range of 100-20,000.  Is this not correct? How do you handle the raw TEROS EC data?

peterregier, 2024-04-01:
Hi Ben, I think there might be 2 things going on here. First, dS to uS is 100,000, but m to cm is 1/100, so I think the conversion should be 1000. Second, I'm not positive, but I think Roy may convert values from the TEROS in the logger code, thought that's a question for him, thx!

bpbond, 2024-04-01:
@roylrich ☝️  also fyi @stephpenn1

bpbond, 2024-04-02:
Thanks for checking! Agreed, it doesn't look like any conversion is happening there.

@peterregier If the conversion is 1e3 instead of 1e5,  we'll end up with 90% of values in the range 0-342,000.

[image]

peterregier, 2024-04-02:
I agree with Steph that having Roy weigh in would be helpful. For context, seawater with salinity of 35 PSU has a conductivity of ~50,000 uS/cm, and water added to the estuarine plot at least for TEMPEST 1 was in the range of <10 PSU (so expecting absolute max of < 20,000 uS/cm even if the sensor was measuring water directly).

bpbond, 2024-04-04:
Roy and I met, double-checked everything from sensors through to outputs, and found the issue. Opening a PR shortly to fix. Thank you @peterregier !

stephpenn1, 2024-04-04:
THANK YOU @peterregier for catching this

peterregier, 2024-04-04:
🫡 thank Y'ALL for making this pipedream a pipe-reality!

closed 2024-04-04 by bpbond

stephpenn1, 2024-04-04:
If we ever needed a slogan it'd be this one

bpbond, 2024-04-04:
Fyi we are planning to release v1-0 by the end of next week. That will include this and many other fixes.

PR #130 (2024-04-04): TEROS12 EC data are already in µS/cm
Fixes #119