COMPASS-DOE/sensor-data-pipeline #127: The Which column

bpbond, 2024-04-04:
@stephpenn1 

| Example | "Instrument" | "Which" | "Individual" | Notes |
|--------|--------|--------|--------|--------|
|  Aquatroll | Model (200 or 600) | Number/name within plot | `NA` | OWC UP and TR have A, B, C trolls
| Sapflow | "Sapflow" | `NA` | Tree number |
| TEROS | "TEROS12" | `NA` | Probe number |
| Check (datalogger) | "Datalogger" | Datalogger number | `NA` |
| Exo | "EXO" | `NA` | `NA` | 
| ClimaVue | "ClimaVUE50" | `NA` | `NA` | 

Following up on our discussion yesterday, here's a table showing how each of the L1 fields is used for each data type.

Right now, **it's OWC trolls that make us need a `Which` column**.

stephpenn1, 2024-04-04:
| Example                   | "Instrument"             | Unique_ID                     | Location       | Unique_ID Description  |
|--------|--------|--------|--------|--------|
|  Aquatroll                 | Model (200 or 600) | SN                                   | Plot                   | Serial Number
| Sapflow                    | "Sapflow"                 |         ??                             | Tree number    | Port Number?
| TEROS                      | "TEROS12"               | Probe number               | Grid Cell           | Sensor ID
| Check (datalogger) | "Datalogger"            | Logger Number             | Plot                   | Logger Number
| Exo                            | "EXO"                       | SN                                   | Plot                   | Serial Number
| ClimaVue                  | "ClimaVUE50"          | SN                                  | Site                    | Serial Number

** _Note in this example, we would need a mapping document that joins with Unique_ID to add in Location_ separate from design_table

*** _Second Note: This will introduce repeats in long format, so we'd have to access data volume increase_

bpbond, 2024-04-05:
NICE -- definitely getting closer! But I am concerned about the high data repetition; those serial numbers could kill us.

What about:

| Example                   | "Instrument"             | "Instrument_ID"           | "Location"    |
|--------------------|---------------------|-----------------------|--------------|
|  Aquatroll                 | "AquaTROLL200" or "AquaTROLL600" | "A"/"B"/"C"       | Grid cell (TEMPEST) or "Plot" (synoptics)      |
| Sapflow                    | "Sapflow"                 |         `NA`                             | Tree number    |
| TEROS                      | "TEROS12"               | Probe number               | Grid cell (TEMPEST) or "Plot" (synoptics)        |
| Check (datalogger) | "Datalogger"            | Logger number             | "Plot"                   |
| Exo                            | "EXO"                       | `NA`                                   | "Plot"                   |
| ClimaVue                  | "ClimaVUE50"          | `NA`                                | "Site"                    |

👉 Data release will include mapping file for tree number -> grid number and species

bpbond, 2024-04-08:
Minor question: in L1 we're using "NA" for missing values. Keep this? We could also change to "" (empty string).

stephpenn1, 2024-04-08:
> NICE -- definitely getting closer! But I am concerned about the high data repetition; those serial numbers could kill us.

I think this exactly is great

> Minor question

Will R and other programming languages read a blank cell as NA automatically? I want to say yes. Second question, if we had "NA" here, would that add to the memory?

I want to there's a difference between `column = NA` vs. `column = "NA"` right?

bpbond, 2024-04-08:
>I want to say there's a difference between column = NA vs. column = "NA" right?

Yes! The first sets to `NA`, and the second sets to a string "NA".

So R (but not many other languages) has an explicit NA, right. What I'm wondering is when we write the L1 data out as csv files, what are NA values written as? This is `write.csv`'s  `na` parameter:
[image]

Files written with "NA" for missing values versus "" will be slightly bigger, yes.

Okay, anyway, thank you. I will plan to produce the draft L1 dataset following the schema above, and then we can evaluate, if that sounds good.

stephpenn1, 2024-04-08:
Sounds great - thank you!

closed 2024-04-10 by bpbond