COMPASS-DOE/sensor-data-pipeline #144: v0.9 L1 TEMPEST data only has sapflow and teros

stephpenn1, 2024-04-09:
Looks like we are only writing out TMP data for teros and sapflow. The design_table does have research names for aquatroll, but they are not included in the final.

Taking a look at a data file:
[image]

Produces:
[image]

And pdf summary:
[image]

bpbond, 2024-04-09:
This is currently still true for v1-0 😞 
```r
 january_2024_data %>% filter(!is.na(value)) %>%  group_by(Plot, Instrument) %>% summarise(n = n()) %>% pivot_wider(values_from = "n", names_from = "Plot")
`summarise()` has grouped output by 'Plot'. You can override using the `.groups` argument.
# A tibble: 2 × 4
  Instrument      C      F      S
  <chr>       <int>  <int>  <int>
1 Sapflow     71262  77370  77358
2 TEROS12    400065 409617 408903
```

bpbond, 2024-04-09:
Ah thank you @stephpenn1 -- I see the problem 🎉  -- missing entry in our bounds/units table.

PR #145 (2024-04-09): Correct research_name entry; change from warning to error
We had an incorrect `research_name` entry in the metadata, which meant that ALL Aquatroll data were being tossed out. This PR
* Fixes the entry
* Changes the code so that a missing entry will now cause a `stop` error, not a warning (which was being missed)

Thank you @stephpenn1 @wilsonsj100 !

Closes #129 
Closes #144
closed 2024-04-09 by bpbond