COMPASS-DOE/sensor-data-pipeline #154: Design Link Removal

wilsonsj100, 2024-04-15:
Hi Data Team, 

Does the beta version take out the design link that tells you which zone the synoptic site data are from? I guess we may be able to link it to the unique ID, but I was really liking having the design_link that told me if a sensor was in TR or WC and what depth for a teros (for example). Maybe I am just missing where this info is now. 

Thanks! 
-Steph

bpbond, 2024-04-15:
Hi @wilsonsj100 

The information should all still be there, just split into different columns: Plot, Sensor_ID, research_name, etc.

wilsonsj100, 2024-04-15:
[image]
Hi @bpbond

hmmm okay... maybe I am missing where that is. Here is an MSM file in the v1-0 folder for june 2023 after I download it. I am not seeing those other columns?

bpbond, 2024-04-15:
Plot is on the far left, column A; sensor depth will be part of the research_name in H.

wilsonsj100, 2024-04-15:
Whoops.... My bad when I subset the data I took the original column names from v0-9 and cut that column out! Thank you!

closed 2024-04-15 by bpbond