COMPASS-DOE/sensor-data-pipeline #155: gw_depth missing from L1

bpbond, 2024-04-15:
h/t @wilsonsj100

bpbond, 2024-04-15:
It's not marked as to be included in the output! In addition, no units are given in the variable table.

This is the `Depth600A` AquaTROLL variable, is that correct? What are the units?

bpbond, 2024-04-15:
For reference, here are the troll variables we ARE currently outputting:

battery_voltage
gw_bar_pressure
gw_temperature
gw_salinity
gw_density
gw_ph
gw_ph_orp
gw_rdo_conc
gw_pressure

bpbond, 2024-04-17:
I talked with @peterregier and this omission is intentional, because it needs to be corrected for various factors; ie the raw Aquatroll number is not correct. 

BUT as I sit here typing this, if the corrected value will be in L2, don't we still want it in L1 (but with a warning not to use raw value)?

@stephpenn1 @peterregier

wilsonsj100, 2024-04-18:
Hmmmm I feel like water depth is one that we would probably want to have in L1, but I understand the original intention.

stephpenn1, 2024-04-18:
I don't believe `Depth600A` is used to calculate water level, just looking at a snippet of PRs code:
[image]
It uses pressure and density from the sonde (weather station baro pressure for the 200s), and a static field-measured distance of the sonde line from an inventory sheet. 

I'd be fine with keeping that `Depth600A` variable if it's useful as long as we rename to something more descriptive. Also talking with PR today I was curious: if we calculated wl_below_surface_m and plotted it with Depth would it'd be correlated? I assume somewhat

bpbond, 2024-04-19:
Great detective work and reasoning!

@peterregier @Fausto2504 It would be great to get your expert opinions:

1. Correct that `Depth600A` is not part of the depth calculation, per @stephpenn1 above?
2. Does it have _any_ potential use? Or should we (continue to) drop it?

Thanks for everyone's patience and help to resolve this!

peterregier, 2024-04-19:
> Check twice before you click! This email originated from outside PNNL.

Hi folks,

  1.  Correct, we don’t use Depth600A
  2.  I’m for continuing to drop it completely. My worry is, no matter what caveats we put on it, we know that Depth600A is wrong, but it says depth in the name, and so including it at any step means folks might find it and erroneously use it. I’ve done similar things with other folks’ datasets in the past, since I’m usually too excited to look at data to read the metadata…

Interested in Fausto’s thoughts on this.

Cheers,
Peter
_______________________________
Research Scientist
Coastal Sciences Division
Energy and Environment Directorate
Pacific Northwest National Laboratory

Fausto2504, 2024-04-19:
Hi all,

I am doing fieldwork at OWC and can discuss the possibility of use depth600 next week.

But for now, I agree in deleting depth600 because it is incorrect as it is now and we do not use to calculate wl_below_surface.

Thanks for reaching out!
Best,
Fausto

bpbond, 2024-04-19:
Okay I think if both hydrologists are saying "no don't use this" then we have our answer! 🙏

closed 2024-04-19 by bpbond

Fausto2504, 2024-04-22:
[like]  Machado da silva, Fausto reacted to your message: