COMPASS-DOE/sensor-data-pipeline #159: Weather station data in L1 hourly?

wilsonsj100, 2024-05-01:
@bpbond and I were pulling out some weather station L1 data yesterday and noticed that the L1 datafile only had hourly measurements (at least for air temperatures). I opened a raw SWH file and it looks like the data comes in at 15min intervals? Just wanted to check that it was intentional to take the hourly data for L1. 

Thank you!

bpbond, 2024-05-06:
Checking things

These sites seem to have correct 15-minute weather data
* CRC
* MSM
* OWC
* PTR
* SWH

These sites seem to have data only on the hour
* GWI

bpbond, 2024-05-06:
Compass_SWH_W_421_ClimaVue50_15min_20240502000445.dat:

```
"TOA5","Compass_SWH_W_421","CR1000X","39603","CR1000X.Std.06.01","CPU:COMPASS_v3.3.CR1X","29937","ClimaVue50_15min"
"TIMESTAMP","RECORD","SlrFD_W_Avg","SlrTF_MJ_Tot","Rain_mm_Tot","WS_ms_S_WVT","WindDir_D1_WVT","WindDir_SD1_WVT","MaxWS_ms_Max","MaxWS_ms_TMx","Invalid_Wind_Tot","AirT_C_Avg","AirT_C_Max","AirT_C_TMx","AirT_C_Min","AirT_C_TMn","VP_mbar_Avg","BP_mbar","BP_mbar_Max","BP_mbar_TMx","BP_mbar_Min","BP_mbar_TMn","RH","RHT_C_Avg","TiltNS_deg_Avg","TiltWE_deg_Avg","Strikes_Tot","Dist_km_Min","Dist_km_TMn","PAR_Den_C_Avg","PAR_Tot_C_Tot","CVMeta"
"TS","RN","W/m^2","MJ/m^2","mm","meters/second","Deg","Deg","meters/second","meters/second","","Deg C","Deg C","Deg C","Deg C","Deg C","mbar","mbar","mbar","mbar","mbar","mbar","%","Deg C","degrees","degrees","count","kilometers","kilometers","micromols/s/m^2","mmol/m^2second",""
"","","Avg","Tot","Tot","WVc","WVc","WVc","Max","TMx","Tot","Avg","Max","TMx","Min","TMn","Avg","Smp","Max","TMx","Min","TMn","Smp","Avg","Avg","Avg","Tot","Min","TMn","Avg","Tot","Smp"
"2024-04-02 00:15:00",37088,0,0,0.425,2.175,264.3,22.33,3.89,"2024-04-02 00:09:00",0,11.09,11.2,"2024-04-02 00:01:00",10.9,"2024-04-02 00:14:00",12.51333,1015.1,1015.7,"2024-04-02 00:01:00",1014.9,"2024-04-02 00:09:00",95.9,11.02,1.093,0.32,2,0,"2024-04-02 00:01:00",0,0,"013CAMPBELLCLIM50533VUE-500003414"
"2024-04-02 00:30:00",37089,0,0,0,1.673,294.9,11.28,2.92,"2024-04-02 00:25:00",0,10.94,11,"2024-04-02 00:19:00",10.9,"2024-04-02 00:16:00",12.48667,1014.8,1015.3,"2024-04-02 00:21:00",1014.8,"2024-04-02 00:28:00",95.6,10.93,1.16,0.24,4,0,"2024-04-02 00:16:00",0,0,"013CAMPBELLCLIM50533VUE-500003414"
"2024-04-02 00:45:00",37090,0,0,0,1.825,293.6,5.781,2.87,"2024-04-02 00:39:00",0,10.92,11.1,"2024-04-02 00:43:00",10.8,"2024-04-02 00:33:00",12.42667,1014.4,1014.8,"2024-04-02 00:31:00",1014.4,"2024-04-02 00:45:00",93.7,10.87,1.073,0.24,0,0,"2024-04-02 00:31:00",0,0,"013CAMPBELLCLIM50533VUE-500003414"
"2024-04-02 01:00:00",37091,0,0,0,2.201,284.6,8.13,3.51,"2024-04-02 00:46:00",0,11.01,11.1,"2024-04-02 00:51:00",11,"2024-04-02 00:46:00",12.36667,1013.8,1014.4,"2024-04-02 00:46:00",1013.8,"2024-04-02 01:00:00",93.8,10.79,1.067,0.24,0,0,"2024-04-02 00:46:00",0,0,"013CAMPBELLCLIM50533VUE-500003414"
```

Resulting L1 data:

[image]

[image]

bpbond, 2024-05-06:
About GWI, here are the raw contents from `Compass_GWI_W_411_ClimaVue50_15min_20240502000154.dat`:

```
"TOA5","Compass_GWI_W_411","CR1000X","28726","CR1000X.Std.05.01","CPU:COMPASS_v3.CR1X","57162","ClimaVue50_15min"
"TIMESTAMP","RECORD","SlrFD_W_Avg","SlrTF_MJ_Tot","Rain_mm_Tot","WS_ms_S_WVT","WindDir_D1_WVT","WindDir_SD1_WVT","MaxWS_ms_Max","MaxWS_ms_TMx","Invalid_Wind_Tot","AirT_C_Avg","AirT_C_Max","AirT_C_TMx","AirT_C_Min","AirT_C_TMn","VP_mbar_Avg","BP_mbar","BP_mbar_Max","BP_mbar_TMx","BP_mbar_Min","BP_mbar_TMn","RH","RHT_C_Avg","TiltNS_deg_Avg","TiltWE_deg_Avg","Strikes_Tot","Dist_km_Min","Dist_km_TMn","PAR_Den_C_Avg","PAR_Tot_C_Tot","CVMeta"
"TS","RN","W/m^2","MJ/m^2","mm","meters/second","Deg","Deg","meters/second","meters/second","","Deg C","Deg C","Deg C","Deg C","Deg C","mbar","mbar","mbar","mbar","mbar","mbar","%","Deg C","degrees","degrees","count","kilometers","kilometers","micromols/s/m^2","mmol/m^2second",""
"","","Avg","Tot","Tot","WVc","WVc","WVc","Max","TMx","Tot","Avg","Max","TMx","Min","TMn","Avg","Smp","Max","TMx","Min","TMn","Smp","Avg","Avg","Avg","Tot","Min","TMn","Avg","Tot","Smp"
"2024-03-02 04:00:00",15258,0,0,0,3.306,132.7,9.37,7.21,"2024-03-02 03:48:00",0,9.46,9.5,"2024-03-02 03:39:00",9.4,"2024-03-02 03:33:00",11.75,1025.5,1026.6,"2024-03-02 03:34:00",1025.5,"2024-03-02 04:00:00",100,9.42,-2.807,0.936,0,0,"2024-03-02 03:33:00",0,0,"013CAMPBELLCLIM50529VUE-500002163"
"2024-03-02 05:00:00",15259,0,0,0,3.773,131.7,9.78,9.18,"2024-03-02 04:47:00",0,9.84,10.1,"2024-03-02 04:41:00",9.5,"2024-03-02 04:01:00",12.12333,1024.7,1025.6,"2024-03-02 04:01:00",1024.6,"2024-03-02 04:49:00",99.7,9.75,-2.837,0.79,0,0,"2024-03-02 04:01:00",0,0,"013CAMPBELLCLIM50529VUE-500002163"
"2024-03-02 06:00:00",15260,0,0,0,4.932,132.9,9.14,12.01,"2024-03-02 05:24:00",0,10.41,10.7,"2024-03-02 05:54:00",10,"2024-03-02 05:01:00",12.58667,1023.2,1024.6,"2024-03-02 05:02:00",1022.9,"2024-03-02 05:49:00",100,10.32,-2.998,1.057,0,0,"2024-03-02 05:01:00",0,0,"013CAMPBELLCLIM50529VUE-500002163"
"2024-03-02 07:00:00",15261,0.267,8E-05,0,3.869,128.6,2.848,9.66,"2024-03-02 06:04:00",55,10.84,11,"2024-03-02 06:43:00",10.6,"2024-03-02 06:08:00",12.965,1021.7,1023.3,"2024-03-02 06:02:00",1021.7,"2024-03-02 07:00:00",99.9,10.78,-2.932,0.758,0,0,"2024-03-02 06:01:00",1.045,3.763,"013CAMPBELLCLIM50529VUE-500002163"
```

Note above that this table, the 15-minute climate data, **is only being output every hour**.

So right now I see what look like a GWI datalogger issue @wilsonsj100 -- but not (at least not yet) a processing-code issue.

wilsonsj100, 2024-05-07:
Hey Ben, 
Thank you for checking this. Did you check all the files for the 15min thing? I have an MSM file that is only hourly as well that I have been using for building out my airtemps data. 
[Uploading MSM_20220601-20220630_L1_v1-0.zip…]()

wilsonsj100, 2024-05-07:
@bpbond  I realized I didn't tag you in the post above. That file was part of the beta release, but I checked a new L1 MSM file from 2022 and saw the same issue with only hourly weather data. Not sure there is anything to do here, just notifying.

bpbond, 2024-05-07:
Thanks @wilsonsj100 . It looks like the MSM marsh datalogger was only outputting hourly ClimaVUE50 data at first; in January 11 2023 it switched to 15 minute frequency.

bpbond, 2024-05-08:
As before, I don't see any processing pipeline issues here, just dataloggers that were/are outputting hourly. So closing. But Steph feel free to reopen if that's wrong. 

Also I'm happy to help with some extrapolate-to-15-minutes code if useful.

closed 2024-05-08 by bpbond