COMPASS-DOE/sensor-data-pipeline #160: Flipped solar unit in design table

bpbond, 2024-05-01:
Following up on #158 , from our variable metadata file:

[image]

But...
W/m2 = J/s/m2 is a _rate_
MJ/m2 = _amount_ of energy

So it seems like the "total solar flux" should be in MJ/m2, and the "average" should be W/m2...are the units flipped for these entries?

💡  Ah the variable names have the correct units! OK, so I think our entries are wrong.

@stephpenn1 @roylrich @selinalcheng Is this right? I mean, is it right that the metadata table is wrong? Thanks

selinalcheng, 2024-05-07:
Yes, I think variable names/units are correct -- you can look at variable names to confirm.

bpbond, 2024-05-07:
Sorry, does that mean you agree that the COMPASS-FME units are flipped by accident?

selinalcheng, 2024-05-07:
Yes, I think the metadata units are flipped by accident and the units hard-coded into the loggernet variable are correct.

closed 2024-05-08 by bpbond