COMPASS-DOE/sensor-data-pipeline #169: Request to retain dedicated plot column for TMP L1

peterregier, 2024-05-07:
I just pulled L1 data and it appears the plot column included in prior releases is now removed and instead in the filename. I recognize and appreciate trying to reduce redundancy and am happy to parse from filenames, but as critical metadata all TEMPEST analyses will need, I think it warrants a dedicated column for ease of use. I'm intending this comment to only apply to TEMPEST.

bpbond, 2024-05-07:
Thanks @peterregier 

Why it is "only apply to TEMPEST"? Plot is a crucial piece of information everywhere else too, right? What about Site? Should that have its own dedicated column? Curious about your thoughts here.

>for ease of use

That's definitely a priority, so I'm very open to this. Just would like to understand better why something like this seems onerous:
```r
# Read in all the 2024 TEMPEST control plot data
f <- list.files("TMP_2024/", pattern = "TMP_C_", full.names = TRUE)
dat <- bind_rows(lapply(f, read_csv))
dat$Site <- "TMP"
dat$Plot <- "C"
```

bpbond, 2024-05-07:
Just making a note, adding a "Site" and "Plot" column to one of the 2024 TEMPEST files (I tested `TMP_C_20240301-20240331_L1_v1-0.csv`) increases its size from 44 to 47.3 MB. Not bad.

peterregier, 2024-05-07:
@bpbond my instinct on "easier" comes from my solution of parsing filenames with stringr::str_extract(name, "(?<=_)[^_]+(?=_)") when reading all csvs from a given L1 folder (eg TMP_2023). Absolutely agree your solution is not onerous, just hoping to keep the data as easy as possible for folks of all to use, if folks don't have to parse strings or assign, that's one less barrier in my mind. 

I think you're absolutely right, having transect location as an equivalent variable would make sense for synoptics! Adding site as a column makes sense to me too. I recognize wanting to keep things lightweight, just putting in my 2 cents on that balance.

bpbond, 2024-05-07:
It does seem penny-wise, pound-foolish to save 10% in file sizes but force everyone to again and again parse filenames to re-create those `Site` and `Plot` columns. 

Thoughts @selinalcheng @roylrich @wilsonsj100 ?

bpbond, 2024-05-08:
Addressed in #167

wilsonsj100, 2024-05-08:
Sorry for the late input, but I am pro adding the zone back in! Thank you!

roylrich, 2024-05-08:
agreed

Roy Rich PhD.

Research Ecologist

Smithsonian Environmental Research Center

647 Contees Wharf Rd.

Edgewater, MD USA 21037

Cell 651-328-4391

***@***.******@***.***>

https://serc.si.edu/technology-in-ecology

External Email - Exercise Caution