COMPASS-DOE/sensor-data-pipeline #178: Synoptic dashboard erroring out

stephpenn1, 2024-05-31:
[image]

bpbond, 2024-05-31:
Thanks @stephpenn1 . Note the new TEMPEST dashboard is at https://github.com/COMPASS-DOE/TEMPESTdash -- were you running that one, or the old one here?

stephpenn1, 2024-05-31:
This is for the synoptic site dashboard - different than TEMPESTdash

stephpenn1, 2024-05-31:
Synoptic dash is in this repo located here: https://github.com/COMPASS-DOE/data-workflows/tree/main/synoptic_dashboard

bpbond, 2024-05-31:
Ah, sorry!

stephpenn1, 2024-05-31:
https://compassdashboards.shinyapps.io/synoptic_dashboard/

Also note this is low priority, I'm not sure if people still find this dashboard useful

stephpenn1, 2024-07-17:
I've refreshed the drop token and this is back up and running

closed 2024-07-17 by stephpenn1