COMPASS-DOE/sensor-data-pipeline #182: Are sonde data being handled correctly?

bpbond, 2024-06-25:
Etienne notes that sonde data are almost always blank in v1-0, which is weird. I responded

>Agreed, from some quick spot-checking, it looks like sonde data are almost always blank. An exception is PTR in 2022 (attached). Do we have an example of sondes that we should have data for? I’m trying to find a raw file with data to trace it and see if we have a systematic problem or not.

It would be good to check this and talk with Fausto/others to make sure we're not accidentally dropping sonde data.

bpbond, 2024-07-19:
FYI @stephpenn1  - I don't think we have the bandwidth to deal with this in the next week, so removing it from the `v1-1` milestone.

stephpenn1, 2024-07-19:
Fine with me

bpbond, 2024-12-31:
| Site | L1 sonde data | Raw data status |
|--------|--------|--------|
| CRC_2024 | Rows present but no data | Sporadic raw data: OK through 2022-10-21, then NAN to 2023-04-21, data to 2023-11-17, NAN through 2024-04-30; NAN again after 2024-06-02. **When there's raw data, there's L1 data** |
| GCW_2024 | No rows (as expected) | Cell |
| GWI_2024 | Rows present but no data | Cell |
| MSM_2024 | Rows present but no data | Cell |
| OWC_2024 | Data present 👍  | Cell | 
| PTR_2022 | Data present but plot is "Buoy" | Cell |
| PTR_2024 | Rows present but no data | Cell |
| SWH_2024 | Rows present but no data | Cell |
| TMP_2024 | No rows (as expected) | Cell |

bpbond, 2024-12-31:
Thanks for flagging this issue @efluet and sorry it took me so long to come back to it.

After some detailed investigation: yeah, we _definitely_ have a problem here.
* The minor issue is that for MSM and PTR in 2022, the sonde data are labeled as plot "Buoy", not plot "W", which is confusing but an easy fix.
* The major issue is that, except for MSM and OWC (see summary table above), **almost all of the raw `ExoTable` files have nothing but NANs in them**. This is potentially a very large data loss! 

Tagging @roylrich -- do you know about this?

Tagging @Fausto2504 @stephpenn1 for their awareness/feedback, thanks

bpbond, 2024-12-31:
>the sonde data are labeled as plot "Buoy", not plot "W", which is confusing but an easy fix.

Hmm, _should_ sonde data be grouped with "W" data? Or do they stay as they own "Buoy" plot? 

@stephpenn1 ?

roylrich, 2024-12-31:
Ok a few things that I think are going on:
1. SWH and GWI don't have sondes. The data should be taken from nearest NERRS sites
2. GCW site should come from GCREW marsh outlet data. @ewestbro in TElab is in charge of these data
3. MSM is on a separate logger from W in Buoy
4. PTR has odd history as it was on buoy for one year and then Matt and Fausto swaped to posted site. I am sure that data files from sondes need to be integrated with our LI data set.
5. CRC situation is no know to me but probably has EXO data files that need to be incorporated

We spoke in last data meeting that we would need to pull these data and integrated into datastream but also need to check time offsets on KOR EXO data. 
We need to get NERRS data as well.

roylrich, 2024-12-31:
Just posted on site
Ok a few things that I think are going on:

  1.  SWH and GWI don't have sondes. The data should be taken from nearest NERRS sites
  2.  GCW site should come from GCREW marsh outlet data. @ewestbro<https://github.com/ewestbro> in TElab is in charge of these data
  3.  MSM is on a separate logger from W in Buoy
  4.  PTR has odd history as it was on buoy for one year and then Matt and Fausto swaped to posted site. I am sure that data files from sondes need to be integrated with our LI data set.
  5.  CRC situation is no know to me but probably has EXO data files that need to be incorporated

We spoke in last data meeting that we would need to pull these data and integrated into datastream but also need to check time offsets on KOR EXO data.
We need to get NERRS data as well.

Roy Rich PhD.

Research Ecologist

Smithsonian Environmental Research Center

647 Contees Wharf Rd.

Edgewater, MD USA 21037

Cell 651-328-4391

***@***.******@***.***>

https://serc.si.edu/technology-in-ecology

External Email - Exercise Caution

Hmm, should sonde data be grouped with "W" data? Or do they stay as their own "Buoy" plot?

@stephpenn1<https://github.com/stephpenn1> ?

bpbond, 2024-12-31:
Thank you @roylrich that's super helpful. OK so we need to

- [ ] Remove SWH and GWI EXO from design table so those empty rows don't appear in L1
- [ ] If MSM marsh in a separate logger, maybe that answers my question above about plot name for sonde data? Will let Steph make that call
- [ ] PTR and CRC seem complicated and may need to be broken into separate issues. I just checked the current Dropbox and CRC has good EXO data but PTR only NANs. Integrating data files will need #213 and thus `v2-0`

stephpenn1, 2025-01-02:
> > the sonde data are labeled as plot "Buoy", not plot "W", which is confusing but an easy fix.
> 
> Hmm, _should_ sonde data be grouped with "W" data? Or do they stay as their own "Buoy" plot?
> 
> @stephpenn1 ?

I say separate, honestly open water is the end transect location imo and should have it's own plot naming convention (e.g. if we ever deploy other sensors out there in addition to the buoy). What about something like "OW"?

roylrich, 2025-01-02:
When we do Spring program update, we can push version that only has exotable where they are actually deployed. This will resolve the false positive data tables coming in. We will need to fix naming. The OW (open water) solution seems correct approach. _RR

Roy Rich PhD.

Research Ecologist

Smithsonian Environmental Research Center

647 Contees Wharf Rd.

Edgewater, MD USA 21037

Cell 651-328-4391

***@***.******@***.***>

https://serc.si.edu/technology-in-ecology

stephpenn1, 2025-01-02:
Is that more of a 2-0 change then? This reminds me that we should set our criteria for releases!

roylrich, 2025-01-02:
I think it is a 2-0 change and we should just get going on that... I am not sure we need a 1-2 change

Roy Rich PhD.

Research Ecologist

Smithsonian Environmental Research Center

647 Contees Wharf Rd.

Edgewater, MD USA 21037

Cell 651-328-4391

***@***.******@***.***>

https://serc.si.edu/technology-in-ecology

bpbond, 2025-01-02:
I don't know--it all depends on criteria. 

I guess my vote is, yes, it can be in v1-2. Because, while it could break scripts in theory, probably it won't (since not many people are using sonde data yet afaik).

bpbond, 2025-01-02:
Hahaha it's your call @stephpenn1

Fausto2504, 2025-01-02:
Thanks, Ben and everyone, for the discussion and for including me! LE sites are labeled as 'W' and MSM as 'Buoy.' Honestly, I’m not a fan of either. We already use 'W' for wetlands/muddy edges, and 'Buoy' doesn’t fit since our Exos aren’t in buoys (except PTR in 2022). How about 'OW' for open water instead? Thoughts?

bpbond, 2025-01-02:
Thanks @Fausto2504 

Agreed, @stephpenn1 's "OW" wins the day, and will make things clearer and more consistent. The only question is [whether it will be part of the v1-2 release this month, or have to wait for v2-0](https://github.com/COMPASS-DOE/sensor-data-pipeline/issues/226).

Fausto2504, 2025-01-02:
I’ll have the full set of LE sites from March 2022 to December 2024 ready soon. There were issues with data loggers affecting LE data at times, but I’ve been doing thorough QAQC. I can provide both the cleaned internal data, and the codes used for flagging errors.

bpbond, 2025-01-02:
@Fausto2504 That's great! Thank you for your work there. Integrating those directly downloaded data will _definitely_ be a v2-0 thing, but we've already worked out the basic framework--see #216

bpbond, 2025-02-25:
I think this has been obviated by #216 #213 and the already-done #241

closed 2025-02-25 by bpbond

bpbond, 2025-02-25:
Yes, this information was added in 2feaefe59735c752529be80c1bb98753a40f0ccb