COMPASS-DOE/sensor-data-pipeline #186: Freshwater Plot in TMP data frames is Labeled as "FALSE"

amyerspigg, 2024-07-12:
[image]

amyerspigg, 2024-07-12:
This appears to be the case in both of the two years I pulled 2022 and 2023.

bpbond, 2024-07-13:
Thanks for your sharp eyes here! I started to type "omg, something is wrong, I will investigate and fix the code" ...but then realized that we're writing the freshwater plot abbreviation ("F") into that column...and R is thinking that it's a logical (TRUE/FALSE) because of that "F" I bet.

🤦 

How are you reading in this data @amyerspigg ?

amyerspigg, 2024-07-15:
@bpbond & @stephpenn1 I am using the read_csv function

stephpenn1, 2024-07-15:
@bpbond I'm looking into this using her download / read code and will report back

stephpenn1, 2024-07-15:
AMP is using drive_download to access these files, so I tested used (1) her code & that function and (2) direct download from google drive; then I tried using `read_csv()` and `read.csv()` on those files and all get the same result. For files that are all Freshwater plot they are read in as "logical", see below example for the two read functions:
[image]

stephpenn1, 2024-07-15:
For our records, this is linked to https://github.com/COMPASS-DOE/TEMPEST-1-porewater/issues/7

bpbond, 2024-07-15:
Thank you @stephpenn1 

So right, the parser is seeing a column full of "F" characters and is assuming that's logical. @amyerspigg in your workflow, would a solution like this work?
```r
read_csv("TMP_F_20220101-20220131_L1_v1-0. csv", col_types = "ccTcdccdccdd")
```

bpbond, 2024-07-19:
I've tweaked the code of, and added an explanatory comment about this to, our [file-reading example](https://github.com/COMPASS-DOE/data-workflows/blob/main/synoptic/data/L1/code_examples/create-time-series.R) that's included with each release.

I think it's OK to close this issue, and that the code above will be a good solution for you @amyerspigg . But please re-open if needed! Thanks

closed 2024-07-19 by bpbond