COMPASS-DOE/sensor-data-pipeline #188: Redox descriptions

stephpenn1, 2024-07-17:
We need the following information for the redox probes:
- [ ] Descriptions for each depth and reference probe meanings
- [ ] sensor units
- [ ] any conversions needed

sensor_units | final_units | conversion | low_bound | high_bound | description
-- | -- | -- | -- | -- | --
 ? | ? | ? | -1 | 1 | Redox at [ ] cm depth; Reference "a" means [  ], reference "b" means [  ]
 ? | ? | ? | -1 | 1 | Redox at [ ] cm depth; Reference "a" means [  ], reference "b" means [  ]
 ? | ? | ? | -1 | 1 | Redox at [ ] cm depth; Reference "a" means [  ], reference "b" means [  ] 
 ? | ? | ? | -1 | 1 | Redox at [ ] cm depth; Reference "a" means [  ], reference "b" means [  ]

stephpenn1, 2024-07-17:
From @peterregier:
>Sensor units are voltage (I assumed mV but if Roy’s bounds are -1 to 1 then V?). “a” and “b” refer to duplicate reference electrodes used to correct raw redox. They should be reading the same thing but due to the heterogeneity of the system and sensitivity of probes to that environmental complexity, they often read slightly (or sometimes very) different things.  This is a non-trivial component of QC that for me at least that requires looking hard at the data for a given time-period and making a judgement call on what’s “best”. I don’t think there are actual bounds, and these are raw mV, not Eh, so even harder to know, but I agree with Roy: -1 to 1 if V, -1000 to 1000 if mV. Converting to Eh will need to happen, but requires temperature data from TEROS.

bpbond, 2024-07-17:
Channel-to-depth assignments from @peterregier 's code:
```r
set_depths <- function(data){
  data %>% 
    pivot_longer(cols = contains("redox_"), names_to = "sensor", values_to = "redox_mv") %>% 
    separate(sensor, into = c("scrap", "ref", "sensor"), sep = "_") %>% 
    mutate(depth_cm = case_when(sensor == "1" | sensor == "5" | sensor == "9" | sensor == "13" | sensor == "17" ~ 5, 
                                sensor == "2" | sensor == "6" | sensor == "10" | sensor == "14" | sensor == "18" ~ 15, 
                                sensor == "3" | sensor == "7" | sensor == "11" | sensor == "15" | sensor == "19" ~ 30, 
                                sensor == "4" | sensor == "8" | sensor == "12" | sensor == "16" | sensor == "20" ~ 50,
                                TRUE ~ 0)) %>% 
    select(-c(statname, scrap, timestamp))
}
```
Note however from @aestearns01 email that installed depths are actually 10/25/35/55.

bpbond, 2024-07-18:
The redox data seem to be in mV not V:

[image]

FYI @roylrich and I have modified the table above.

stephpenn1, 2024-07-18:
> Channel-to-depth assignments from @peterregier 's code:
>
> ```r
> set_depths <- function(data){
>   data %>%
>     pivot_longer(cols = contains("redox_"), names_to = "sensor", values_to = "redox_mv") [...]

Note the depths are actually 10/**20**/35/55 per Alice's email

I worked with PR to come up with this description: _Redox at [ ] cm depth; Each sensor measures two values based on electrode "A" or "B" reference probes, denoted [redox_A or redox_B; however we plan the name to be]_

roylrich, 2024-07-19:
mV is correct. Thanks

External Email - Exercise Caution

PR #191 (2024-07-19): Infrastructure for TEMPEST redox
Design links, metadata, and test data to add redox

Closes #188 
Closes #162
closed 2024-07-19 by bpbond