COMPASS-DOE/sensor-data-pipeline #193: v1-1 checklist

bpbond, 2024-07-18:
- [ ] Update raw data through end of June
- [ ] Add all old 1min and 5min data to raw
- [ ] Add all old redox data
- [ ] New README

bpbond, 2024-07-18:
Hi @stephpenn1 a brief update -

* As I mentioned yesterday, I was a little concerned about the timeline. But I cleared a things of my calendar and did a bunch of prep work and PRs and am feeling much better
* I just finished a full test run of the v1-1 (data through end of June; 1min and 5min data included; redox included) and it went well. 206 million observations! 
* I'll take a close look at the test run outputs, and we have some things to [nail](https://github.com/COMPASS-DOE/data-workflows/issues/188) [down](https://github.com/COMPASS-DOE/data-workflows/issues/182)
* Overall, however, I think we can produce a final v1-1 by a week from now so that you are able to submit to ESS-DIVE by end of the month

stephpenn1, 2024-07-19:
That is great news, thanks so much for your work on this

bpbond, 2024-07-21:
@stephpenn1 I am traveling the next couple days, doing a little Sunday work so things are ready for you tomorrow.

The v1-1 beta has been uploaded to the [google drive](https://drive.google.com/drive/folders/1_Eq_vc3us40bf_4rNTl1n_Bq7FeVtI0W?usp=drive_link) for you to look through and check. Things for us to do:

- [ ] Resolve questions to Roy in #175 and merge that PR
- [ ] Carefully read the main README
- [ ] I have spot-checked most of the new data (June and July 2024) but please feel free to do so as well
- [ ] Anything else you see

When things look good to you and we have an up-to-date `main` I will regenerate, upload the final v1-1, and let you take it from there (although obv am happy to talk if useful). OK?

PR #203 (2024-08-04): v1-1
Closes #201 
Closes #193
closed 2024-08-04 by bpbond