COMPASS-DOE/sensor-data-pipeline #205: GCW has NA Instrument names

stephpenn1, 2024-08-27:
[image]

[image]

stephpenn1, 2024-08-27:
@bpbond also just realized my x axis is incorrectly ordered, any thoughts on how to fix this?

```
 mutate(
           ts_str = format(date, "%b-%Y"),
           ts_fct = factor(ts_str, levels = unique(ts_str)))
```

bpbond, 2024-08-27:
```r
arrange(date) %>%
mutate(
           ts_str = format(date, "%b-%Y"),
           ts_fct = factor(ts_str, levels = unique(ts_str)))
```

?

stephpenn1, 2024-08-27:
Hmmm no, the dataset is arranged correctly with this but is graphing the same. I just pushed up `availability_graph.R` to `heatmap` branch if you want to take a look at the entire code

bpbond, 2024-08-27:
See PR! I didn't test it on the full dataset though—please lmk if it doesn't fix things

bpbond, 2024-09-03:
Fixed in #206

closed 2024-09-03 by bpbond