COMPASS-DOE/sensor-data-pipeline #209: Wrong depth assigned to TEROS sensor?

bpbond, 2024-09-05:
One of the plots I prepared for #208 is this one:

[image]

Look at the enormous jump that one of the 30cm lines takes -- hmm, do we think this is real? I wonder if it's actually at 5 or 15 cm.

bpbond, 2024-09-05:
@aestearns01 See graph above...any concerns? I checked the "TEMPEST_TEROS_Network_Location&Status" spreadsheet and yes T173 is listed as 30 cm.

 
[image]

bpbond, 2024-12-31:
Coming back to this, with apologies for the delay.

T173 is certainly spiking more strongly than any other 15 cm probe:

[image]

Here's the nest at Control H3:

[image]

Finally, here are all the multi-depth control plot nests in June 2021:
 
[image]

bpbond, 2024-12-31:
I don't see anything conclusive here about a metadata mistake, and while T173 is jumpy, it's not wildly out of line with the others. So closing.

closed 2024-12-31 by bpbond

aestearns01, 2025-03-21:
@bpbond Evan was able to verify in the field that T173 is indeed at 30 cm

bpbond, 2025-03-21:
Thanks for the follow-up @aestearns01 🎉