COMPASS-DOE/sensor-data-pipeline #210: Bad vwc sensor 4 at PTR-W?

bpbond, 2024-09-07:
@robertapeixoto notes that PTR soil moisture data look a little funky in the wetland:

[image]

What's up with sensor 4?

bpbond, 2024-09-10:
Ah, okay, thank you! So there are two issues
* We need to remove the sensor 4 in the wetland — it's not real. This is a quick edit to the design table.
* Why aren't the bad values being removed? First we will check the bounds/units table, then the code.

stephpenn1, 2024-09-10:
Note this is related to #200

I agree with your summary Ben, and we should check what is different about these two sensors vs the other open ports in the other synoptic sites since these show up in the raw data and L1.

bpbond, 2024-09-12:
@robertapeixoto FYI @stephpenn1 and I are both a little jammed up at the moment but will look at this in early October if that's OK.

robertapeixoto, 2024-09-12:
That sounds great to me! Let me know if I can help. =) I had the same out-of-sensor range values issue with ClimaVue data for temperature. So let me know when you get back to this. I might save other graphs here so you know which ones to check in L1 in October. Thanks.

stephpenn1, 2024-10-17:
Hi @robertapeixoto, @bpbond and I looked at this, and have a couple updates:
### Spike
There are data missing from March 8-23,
[image]
and two, when they came back online, VWC went out of bounds (OOB) as you mentioned. We don't remove these in L1 and instead provide the F_OOB columns where 0 is FALSE (in bounds) and 1 in TRUE (out of bounds). the call below will remove bad data
```
data %>% filter(F_OOB == 0)
```
[image]
### TR W sensor 3 and 4
This is an error on our end with how we've set up our experimental design code and will be fixed in the next version release!

For fixing: we need to add a new row to the design table:

Compass_PTR_W_311 | TerosTableC | TerosC({3:4},1) | PTR | W | TEROS12 | C | {3:4} |   |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | --

PR #225 (2025-01-03): Remove nonexistent TEROS 3 and 4 at PTR-W
Closes #210
closed 2025-01-03 by bpbond