COMPASS-DOE/sensor-data-pipeline #227: Do we have coordinates for buoys (OW plots)?

stephpenn1, 2025-01-08:
- [ ] ~SWH~
- [ ] CRC
- [ ] ~GWI~
- [ ] ~GCW~
- [ ] MSM
- [ ] PTR
- [ ] OWC

stephpenn1, 2025-01-15:
Hey @wilsonsj100, im adding site coordinates to any synoptic site that has a OW plot. Do you have any general coordinates for the exo sonde at MSM or OW areas at the CB sites that dont have sondes that you may sample?

wilsonsj100, 2025-01-24:
Hi there, we have general points yes. 

GCReW: 38.87489304517501, -76.54647574933715 (this is the flume and marine geo may have exact coordinates for this one) @roylrich ? 

MSM: 38.428907147561915, -76.23739608746745 the buoy is in beaverdam creek right next to the smithvill rd. bridge and boat ramp. 

SWH & GWI: for these two sites we rely on the CBNERR monitoring sondes, so we do not have L1 OW sites at these locations.