COMPASS-DOE/sensor-data-pipeline #228: AquaTROLL200 conversions + flags

stephpenn1, 2025-01-13:
In _[L1_metadata_variables.csv](https://github.com/COMPASS-DOE/sensor-data-pipeline/blob/main/pipeline/metadata/L1_metadata/L1_metadata_variables.csv)_ I noticed we have information for AquaTROLL600 and not AquaTROLL200 (which are present in the TEMPEST data). Does this mean the 200s aren't getting any conversions or OOB flags?

bpbond, 2025-01-14:
Good question, and I had to go check to be sure of the answer.

* The metadata variables table (formerly called the units and bounds table) is used in `L1_normalize.qmd` -- see [here](https://github.com/COMPASS-DOE/sensor-data-pipeline/blob/cc8946c10c58e133d854a375dd28bdd8cae212af/pipeline/L1_normalize.qmd#L215). 
* But the actual unit conversion step is handled by a function in `compasstools`, and [all it cares about](https://github.com/COMPASS-DOE/compasstools/blob/d48e6f061b709c0207ff8ca8a1720187513a06f7/R/utilities.R#L225) is the `research_name`.
* So, to get to your question, the `table` and `variable` columns in the metadata variables table aren't used by the code at all; they're just context for human readers.
* And you're right, they say AquaTROLL600 only, which is confusing.

stephpenn1, 2025-01-14:
Awesome, thank you for checking!

closed 2025-01-14 by stephpenn1