COMPASS-DOE/sensor-data-pipeline #231: Weird GCW-W TerosTable A and B files

bpbond, 2025-01-24:
In preparation for the new L1 data release I'm testing the pipeline on the last six months of data, and these files unexpectedly popped up (and caused an error):
```
COMPASS_GCW_W_431_TerosTableA_20241101005907.dat
COMPASS_GCW_W_431_TerosTableB_20241101005908.dat
```

In the past we've only had "TerosTableC" at GCW:

[image]

Do you know why on November 1, and no other time, these A and B files got written out @roylrich ? Should we ignore them?

Thanks

bpbond, 2025-01-24:
[image]

roylrich, 2025-01-24:
> External Email - Exercise Caution
>
> Screenshot.2025-01-24.at.10.18.57.AM.png (view on web)<https://github.com/user-attachments/assets/d9cdfcdb-2636-486e-9ac4-08fc3755b8d5>

It is likely that someone turned on the flags for those table at the site. I'm on my way to Brazil. Can we check if the data sets are empty. I'll have Taylor check what flags are on. Thanks

bpbond, 2025-01-24:
>Brazil

Wow! OK. Safe travels and thanks for the quick response.

Confirmed, the files are very shortly and mostly NANs. @stephpenn1 and I will ignore. Thank you!

closed 2025-01-24 by bpbond