COMPASS-DOE/sensor-data-pipeline #233: GWI_W_411 Exotable - empty data, new fields appearing

bpbond, 2025-01-24:
Somewhat similar question to #231 

Two issues. First, the GW_W Exotable data files continue to be full of NANs -- there's no data.

Second, new fields are popping up in them that are erroring the pipeline:

```
! Some data file entries are missing in the design table!
Compass_GWI_W_411 ExoTable TSS_MgL, Compass_GWI_W_411 ExoTable ORP_mv,
Compass_GWI_W_411 ExoTable TDS_mg_L, Compass_GWI_W_411 ExoTable Vertical_Position_m, 
Compass_GWI_W_411 ExoTable Turbidity_FNU, Compass_GWI_W_411 ExoTable Turbidity_Raw
```

We can add these entries easily enough, so that the fields are ignored and dropped, but do we know what's going on here @wilsonsj100 @roylrich ?

wilsonsj100, 2025-01-24:
Hey there Ben,

The COMPASS GWI does NOT have an EXO, so those files should not be there.

We use the CBNERR sonde for that site that is located close by for data.

-Steph

External Email - Exercise Caution

bpbond, 2025-01-24:
Hahahaha well THAT's simple. OK, thank you very much!

closed 2025-01-24 by bpbond

roylrich, 2025-01-24:
The gcw exotable is really the gcrew marsh outlet exotable which is on a different logger. We should turn this flag off as well.

wilsonsj100, 2025-01-24:
[like]  Wilson, Stephanie reacted to your message:

bpbond, 2025-01-24:
Reopening and assigning to @roylrich as a reminder to fix in the future :)