COMPASS-DOE/sensor-data-pipeline #236: MSM "Buoy" and MSM "W" EXO data

bpbond, 2025-01-30:
Since 2022 the MSM datalogger has been generating tables looking like this:

```
Compass_MSM_Buoy_ExoTable_20241231235910.dat
Compass_MSM_W_401_ExoTable_20241231235911.dat
```

This causes problems when we try to map both to @stephpenn1 's new "OW" plot in #229 .

I have checked the raw files extensively and it looks like the "W" files are always NAN, so assume that we can drop the "W" data and map the "Buoy" data to "OW".

Assigning to @roylrich for fixing on the datalogger side!

stephpenn1, 2025-03-25:
#### `03-25-2025`

@roylrich will turn off MSM W ExoTable from the program and any other sites that are producing this file with no sensors plugged in

wilsonsj100, 2025-03-25:
On 3-21-2025 we will also change the time zone for the Buoy, @wilsonsj100 will reach out to @roylrich about this on the day.

bpbond, 2025-04-09:
>On 3-31-2025 we will also change the time zone for the Buoy

🕙 🕚 

Ummm...this seems kind of important. Has it happened @wilsonsj100 ?

roylrich, 2025-04-10:
Yes, it did . TIme change was made : MS buoy reset at 458 on 3/31/205 (558 EDT)

Roy Rich PhD.

Research Ecologist

Smithsonian Environmental Research Center

647 Contees Wharf Rd.

Edgewater, MD USA 21037

Cell 651-328-4391

***@***.******@***.***>

https://serc.si.edu/technology-in-ecology

________________________________
From: Ben Bond-Lamberty
Sent: Wednesday, April 9, 2025 4:57 PM
To: COMPASS-DOE/sensor-data-pipeline
Cc: Rich, Roy; Mention
Subject: Re: [COMPASS-DOE/sensor-data-pipeline] MSM "Buoy" and MSM "W" EXO data (Issue #236)

External Email - Exercise Caution

[https://avatars.githubusercontent.com/u/1956468?s=20&v=4]bpbond left a comment (COMPASS-DOE/sensor-data-pipeline#236)<https://github.com/COMPASS-DOE/sensor-data-pipeline/issues/236#issuecomment-2790981136>

On 3-31-2025 we will also change the time zone for the Buoy

🕙 🕚

Ummm...this seems kind of important. Has it happened @wilsonsj100<https://github.com/wilsonsj100> ?

bpbond, 2025-04-10:
Just to be clear, up until this point it was EDT and now it's EST?

bpbond, 2025-04-10:
Thank you @wilsonsj100 . Okay, it looks like the MSM sonde started logging data in August 2023 and so all data from then until **2025-03-31 05:58 EDT** should have an hour subtracted so final outputs are in EST.

PR #299 (2025-04-11): MSM Buoy EDT timestamp fix
When first installed, the MSM buoy was set to EDT, not EST, and this was only fixed recently per @roylrich [here](https://github.com/COMPASS-DOE/sensor-data-pipeline/issues/236#issuecomment-2792357813). 

This PR back-corrects all the buoy timestamps from August 2023 through March 31 2025. The time series now starts 2023-08-24 15:15:00 EST, equivalent to "2023-08-24 16:15:00" as marked in the original files.

Here's the original raw data at the changeover point (blank line is me snipping out intermediate rows):
[image]

This file starts at 2025-03-31 01:00:00 [EDT] a [... cut]
closed 2025-04-11 by bpbond