COMPASS-DOE/sensor-data-pipeline #242: L1 error - SP to look into

stephpenn1, 2025-02-10:
[image]

bpbond, 2025-02-10:
First question, are you on the `v1-2` branch? That has the necessary changes to the design table.

bpbond, 2025-02-11:
@stephpenn1 `v1-2` didn't have the changes from #229 merged into it. This explains both the errors you were getting and #241

stephpenn1, 2025-02-11:
AH - I'll merge the pull request and run on main thank you

bpbond, 2025-02-11:
The only problem I see is that MSM is still getting a "Buoy" plot because we forgot to change one set of entries in the design table. Pushing a fix.

bpbond, 2025-02-11:
I have updated v1-2 from main, so either way, you should be good to go.

bpbond, 2025-02-11:
Well, in this case I think one of us just overlooked it. Not a big deal!

closed 2025-02-11 by bpbond