COMPASS-DOE/sensor-data-pipeline #244: BBL thinking through v2

bpbond, 2025-02-12:
# Bugs and minor things

* Update deep sapflow names #237

# Data product changes

* Potentially reorganize file structure #171
* Non-datalogger data (which?) #213

# Pipeline improvements

* Better handling of versions, dates, commits

bpbond, 2025-02-12:
>Automated check for missing files

I prototyped a file-count function, below, but this will be tricky, because sometimes files and tables are dropped for good reason. Hmm.

```r
count_files <- function(root = here::here("pipeline/data_TEST")) {
    message("root is ", root)

    count <- function(folder, pattern) {
        f <- list.files(file.path(root, folder), pattern = pattern, recursive = TRUE)
        length(f)
    }

    n_raw <- count("Raw", "\\.dat$")
    n_L0 <- count("L0", "\\.csv$")
    n_L1N <- count("L1_normalize", "\\.csv$")
    n_L1 <- count("L1", "\\.csv$")

    list(Raw = n_raw, L0 = n_L0, L1_normalize = n_L1N, L1 = n_L1)
}
```

stephpenn1, 2025-02-19:
SP Notes: 
* Automated check for missing files at each step of the pipeline (XX files are in Raw but not located in L0, etc.)
* Better status indicator when rendering quarto file
* Restructure branch organization into `main` and `dev` or similar
* List of raw files that should be removed (more documentation on the by hand steps)
* Easy checks we can do - e.g. the data availability graph
* Rethink file structure?

bpbond, 2025-02-19:
* L0 step should check all raw files against (1) a deletion list and (2) a replacement folder
*

stephpenn1, 2025-02-20:
Another thought while the data uploads - do we need to reprocess data every time? Are there certain situations/releases that we don't have to (i.e. only a temporal update)

bpbond, 2025-02-21:
@stephpenn1 Did some editing, reorganization, and sub-issue creation. Feedback welcome

stephpenn1, 2025-02-21:
Possible check - theres a known number of L1 files for those folders that arent the current year (e.g. TMP folders have 73 files on on a full year of data). This could be a quick check

bpbond, 2025-02-21:
Oh that's an interesting idea. Thanks!

bpbond, 2025-06-11:
At this point we are locked in on our v2-0 feature set, so closing this issue. Final checkoff can be handled through the [L1 milestone](https://github.com/COMPASS-DOE/sensor-data-pipeline/milestone/4).

closed 2025-06-11 by bpbond