COMPASS-DOE/sensor-data-pipeline #245: TMP AQ600 density data issue - fix before V2?

wilsonsj100, 2025-02-12:
Found an issue in TEMPEST AQ600 data where the density data. 

This is a scenario where the density is far out of range (0.08 rather than within 0.98-1.05), but the rest of the data during the same time period is reasonable and suggests a working aquatroll. I worked with @peterregier and @roylrich to figure out if this was a sensor, raw data, or pipeline issue by comparing L1 data to loggernet raw data. 

It appears that the sensor was working fine, but the density data was being reported into a different column than it should have been. This shifted the density data over and, therefore, carried an incorrect value through the pipeline. We suspect this is due to settings on the aquatroll changing when a sensor was calibrated, but this is unclear right now. 

We need to identify when these shifts happened throughout the dataset and flag them so that the density data can be reported and carried through to L1. 

This is hard to explain by typing it out, so let me know if there are questions and if I can help in any way! I put an example of one of the L1 pdf's that shows where density data is too low, but the rest of the data from the troll looks reasonable. 
-SWilson

[image]

bpbond, 2025-02-13:
I've gone through all the `gw_density` data, and this is exclusively a problem in TEMPEST Control (34.2% of data) and Fresh (15.6%) plots. It doesn't occur anywhere else.

Here's an example of a raw data file going bad, in which I've snipped out a big middle section of NAN data from 20 April to 3 May 2021:

[image]

When it starts logging again, the first few columns through `Specific_Conductivity600` look good, and maybe `Salinity600`, but then (red line) things seem scrambled. It's not just that the order of columns has changed; some columns don't seem to match up at all.

But the total number of columns is the same, and the last few columns (after red line) look OK:

[image]

bpbond, 2025-02-13:
_Something_ bad clearly happens, right at the same time in April/May 2021, to three of the trolls:

[image]

(Above I said control and fresh only, but the 685410 Aquatroll in salt starts giving a lot of -99999 values at the same time, actually.)

So that tells us the time period. If we can figure out which column is which in the bad data, we should be able to salvage some or much of it.

Hope this is helpful and thanks for the detective work @wilsonsj100 @peterregier @roylrich

wilsonsj100, 2025-02-13:
Hi Ben, 

That is great! Can you call the date that it seems to start? Then I can check the records and see if things were calibrated or updated or something else. 

@roylrich and I looked yesterday and we know which columns in the raw data became switched. @roylrich maybe you can put an example here or paste the column headers? I can't remember the exact column name that density was switched with. Also, we only checked one or two files, so we should double check that it is the same issue throughout the dataset. 

Thanks!

wilsonsj100, 2025-02-13:
I checked our logs with Evan and it looks like the aquatrolls were calibrated in late April of 2021: Control 4-22-2021, Freshwater 4-22-2021, and Saltwater 4-21-2021. The next time they were calibrated was February of 2022 after that.

bpbond, 2025-02-13:
> calibrated in late April of 2021

That lines up perfectly with the example I posted above, which is the earliest example of this behavior.

wilsonsj100, 2025-02-13:
Well that gives us some insight into the mechanism. Seems like after calibration the SDI-12 settings for the Aquatroll's log were entered in a different order than they were in previously, resulting in a column shift. This would have also been reflected on the loggernet current data, but I am not sure that density was a variable being actively monitored there.

bpbond, 2025-02-13:
+1 @wilsonsj100 yes, agreed, that's exactly what I infer.

So at this point, we need:

1. The mapping from 'bad' data to 'good' data
2. Assurance that this problem has been fixed going forward in those three trolls 

Thank you!

bpbond, 2025-02-24:
Just realized that fixing this will tie in nicely with #249 🎉

bpbond, 2025-03-11:
From Evan:
>Hey all, I talked to Roy about this briefly the other day, but the original SDI settings for the TEMPEST aquatrolls had a line for internal temperature, which the newly installed vented aquatrolls do not have, so the bottom 2 lines have been shifted up in the SDI settings. This means that the voltage is reading in the internal temperature line, the battery percent is reading in the voltage line, and the battery percent line is reading as NAN. I was hoping to fix this myself in this round of calibrations but there is no option to add an internal temperature line in the new trolls. The SDI order is the same otherwise. Let me know if there's anything else I can check, I'm assuming this just needs a code adjustment. 

>Yes that's correct Ben, we replaced the old non-vented trolls with vented trolls in September and their SDI settings do not have the line for internal temperature that the non-vented trolls had.

bpbond, 2025-04-10:
@wilsonsj100 Maybe you and I could look hard at this after the GCReW Symposium.

(With Roy and Steph P too if they want of course! But Steph W and I already have time set aside April 24-25)

bpbond, 2025-05-09:
Fixed in #302

PR #302 (2025-05-09): Fix TMP AquaTROLL600 cols
Utility script that is ready to be used to fix #245 

Need to work with @wilsonsj100 to assign column swaps in `utils/TMP_AQ600_bad_column_mapping.csv` for bad files.
closed 2025-05-09 by bpbond