COMPASS-DOE/sensor-data-pipeline #247: Fix gw_salinity bounds

bpbond, 2025-02-13:
@wilsonsj100 said

>Also, I noticed when looking at the bounds in some of the meta-datafiles for TMP that the salinity bounds are listed as 0 to 350 PSU - is that supposed to be 0-35? 

This is `gw_salinity`

bpbond, 2025-02-24:
@wilsonsj100 

Hmm, the [spec sheet for AquaTroll 600](https://in-situ.com/pub/media/support/documents/aquatroll-500-600-700-800-spec-sheet-ltr-en.pdf) says 0-350 PSU is its range. So it seems like our table value is correct...?

bpbond, 2025-02-24:
Yes, absolutely, an L2 flag would incorporate 'unexpected'; but in L1, it's only about sensor specs. 

Thanks lots! Appreciate your sharp eyes.

wilsonsj100, 2025-02-24:
Sorry to raise the alarm on a correct flag setting! I appreciate you checking it out.