COMPASS-DOE/sensor-data-pipeline #249: Editing raw files to fix problems is crappy for reproducibility

bpbond, 2025-02-21:
Editing raw files to fix problems is crappy for reproducibility. We'd prefer to start with raw data and keep track, in the repository, of 'diffs' -- either file deletions or modifications -- that are applied during the L0 step.

1. Reconstruct the current set of file deletions and edits. This involves comparing the Dropbox raw data with the v1-2 `Raw/` folder and figuring out which files were removed; were edited; or (conceivably?) were added.
2. Change L0.qmd to apply the changes as it goes.
3. ...?

bpbond, 2025-02-24:
```
There are 5059 Dropbox raw files
There are 4643 Dropbox raw files
There are 419 Dropbox files but not Raw
...of these 136 have a '_2025' pattern
=====================================================================================
There are 28 Dropbox files whose contents differ from Raw
```
Pretty modest numbers! We have 419-136=283 file removals, and 28 edited files.

bpbond, 2025-02-25:
```
-- Raw/
    | -- Raw_original/
    | -- Raw_edited/
    | -- removed_raw_files.txt
    | -- README.md
```
What about having a structure like this? I.e., the `Raw/` folder contains 
* the original files from Dropbox (`Raw_original/`);
* edited versions of a few that should be used instead of the originals (`Raw_edited/`);
* a list of entirely removed files; and
* the README.

We change `L0.qmd` to process files from `Raw_original/`, skipping files listed in `removed_raw_files.txt`; if a modified version exists in `Raw_edited/`, that is used instead of the original.

bpbond, 2025-02-28:
@stephpenn1 Thoughts? Do you have any feedback on this plan?

stephpenn1, 2025-02-28:
This is a great plan, with `Raw_edited/` I want to make sure I understand the workflow:

_For L0_
- `Raw_edited/` starts off with only the files that needed to be edited (with edits already made and ready to go in the folder)
- As the pipeline runs through `Raw_original/`, it hits a file listed in `removed_raw_files.txt` and pulls from `Raw_edited/` instead
- All files end up in Raw_edited at the end of L0 processing??

bpbond, 2025-02-28:
Ah, good question and sorry for any ambiguity. No, that's not exactly what I'm proposing.

>Raw_edited/ starts off with only the files that needed to be edited (with edits already made and ready to go in the folder)

Correct!

>As the pipeline runs through Raw_original/, it hits a file listed in removed_raw_files.txt and pulls from Raw_edited/ instead

No, if it hits a file listed in `removed_raw_files.txt` then **the file is skipped entirely**.

>All files end up in Raw_edited at the end of L0 processing??

No, Raw_edited only has files added to it by hand, if we determine that a raw file needs to be edited to be successfully processed.

bpbond, 2025-02-28:
I see why my list above was ambiguous. Here's a second attempt:

The new L0 workflow would be:
1. Process files from `Raw_original/`, as before
2. If a file appears is listed in `removed_raw_files.txt`, skip it.
3. Alternatively, if a modified version of the file exists in `Raw_edited/`, process that version of the file.
4. Subsequent steps (reshaping, hash computation, etc.) are the same.

Is that clearer? Thanks!

stephpenn1, 2025-02-28:
Ah okay, so L0 would point to both raw folders now, is that correct?

bpbond, 2025-02-28:
We could implement it several ways.

One way: point L0 to `Raw/` as now, but it 'knows' that it should look inside Raw for the `Raw_original` and `Raw_edited` sub-folders, along with the `removed_raw_files.txt` file.

Another way: pass those folders as parameters to the Quarto file.

stephpenn1, 2025-02-28:
Gotcha, thanks for the explanations. I understand now. That first option sounds great!

bpbond, 2025-02-28:
Awesome. Thank you for the feedback @stephpenn1  — always appreciate your insights.

PR #261 (2025-03-03): Change L0 step to handle removed and edited raw files
This is a big change to the pipeline, following discussion in #249 , so tagging @stephpenn1 for review.

Closes #249
closed 2025-03-03 by bpbond