COMPASS-DOE/sensor-data-pipeline #265: Potential issue when downloading v1-2 data from GDrive

peterregier, 2025-02-28:
Hi, this may be a me problem but wanted to bring up as a potential issue. I am trying to pull all data for TEMPEST 2023 from the v1-2 package. Here's the code I'm using: 

```
## Set directory to pull data from (TEMPEST 2023 data)
data_source = "https://drive.google.com/drive/folders/1ohykyRSfFHdd5eMNmg8miEOreITUpnbc"

## Set location to download data to
local_destination = "data/ignore/v1-2/"

## List the files to download
files <- drive_ls(data_source, pattern = ".csv")

## Loop to download each file
tic("pull files from GDrive")
for(i in 1:nrow(files)){
  drive_download(files$name[i],
                 path = paste0(local_destination, files$name[i]),
                 overwrite = T)
}
toc()
```

This code works, but results in 4 files (FW January, FW March, C Jan, C Nov) now downloading properly. The files show up in my local directory but are all zero bytes. Code did not throw an error. I ended up manually downloading, which worked: 

```
manual_download <- function(i){
  drive_download(files$name[i], 
                 path = paste0(local_destination, files$name[i]), 
                 overwrite = T)
}

manual_download(22)
```

I have two questions: 
1. Is there an issue y'all see in my code that would trigger this / do you have suggestions for safer ways to do this, esp considering it didn't throw an error. 
2. Without an error or warning, do you have suggestions for protecting against this in future (other than double-checking your code's work in the local directory)?

If this is already documented and I didn't see it in existing issues I apologize!

peterregier, 2025-02-28:
[image]

So I did some playing around, and it looks like a OneDrive issue potentially. I changed the directory, reran download, and it chose different files to not download, except now when I open them, there are data. So. I think this is just a PNNL-Mac-OneDrive thing, not a data pipeline or even GDrive issue. Hope y'all have a good weekend and sorry for the false alarm!

closed 2025-02-28 by peterregier

stephpenn1, 2025-02-28:
Ha! Well I'm glad you were able to resolve this and thanks for the detailed GitHub issue anyhow.

Your example code above is a great one, would you be okay with us adding this to a code examples document for others to use? I noticed if you try to download the entire release by hand from the web it jumbles up all the sub folders. This could be a good workaround for those that use the drive for data access.

peterregier, 2025-02-28:
Absolutely! mi código su código, hope it's helpful to someone.

bpbond, 2025-03-03:
Hmm, well, glad it got resolved! Weird.