COMPASS-DOE/sensor-data-pipeline #278: Thinking through Level 2

bpbond, 2025-03-19:
Per VPD emails with Pat, Steph W., Roy:

Compute VPD in the data system using Remko’s [plantecophys package](https://cran.r-project.org/web/packages/plantecophys/index.html) as part of Level 2.

[image]

bpbond, 2025-03-19:
What are key attributes of Level 2 data? And -- crucially -- what's doable with a reasonable amount of effort?

## Easy

* Data normalized to 15-minute and daily and monthly averages
* Perhaps table form

## Medium

* New variables such as VPD (can be computed from existing data)

## Hard

* Cascading flags -- requires a database and significant complexity
* Gap filling
* New variables requiring external data (such as sap flux density)

bsulman, 2025-04-29:
Things on my mind from working with the L1 data so far include converting groundwater pressure to water table level (or surface water level) and converting soil conductivity to salinity if feasible

nickdward, 2025-04-29:
> Things on my mind from working with the L1 data so far include converting groundwater pressure to water table level (or surface water level) and converting soil conductivity to salinity if feasible

Agreed, this is one that would be very valuable and ensure everyone is using the same calculations that consider density effects of salinity, etc., and we already have code for this. I'd vote for groundwater elevation as a high priority. It would be easiest to make this water level belowground surface. Though theoretically we could make it groundwater elevation (NAVD88) with survey data. I could see value in potentially having both for different purposes.

 Likewise, converting soil conductivity to salinity should be straightforward/feasible since we also have temperature from the same sensor.

bpbond, 2025-05-13:
Following the meeting with the Tasks 1 and 3 folks yesterday, minimal L2 features would include

* (Framework for flexibly defining QAQC tests to be performed)
* Group-based outlier identification based on MAD(3) and filtering -- although note [The MAD=0 Problem](https://eurekastatistics.com/using-the-median-absolute-deviation-to-find-outliers/)
*  Average to timestep (say 15 min)
* `complete` for continuous coverage
* Framework for computation of new variables

bpbond, 2025-05-13:
Default time resolution of L2: hourly?

nickdward, 2025-05-13:
> Default time resolution of L2: hourly?

Unless there's a data volume/file size limitation, I would vote for 15 min time steps. In the wetlands and transitions, hourly time steps would frequently miss peak water levels during a flood or high tide, for example. Hourly is probably perfectly fine for the upland systems, but limiting for tidally influenced systems. Here's an example of water level at SWH:

[image]

And from Moneystump, hourly may miss that high high point followed by a more gradual decrease. These are short windows, but in some groundwater modeling I've seen that even a very short exposure to seawater even only a few times a year can greatly influence groundwater salinity:

[image]

bpbond, 2025-05-21:
Effects of different outlier settings (the threshold for identification and the time period data are grouped by) for SWH soil temperature data. Here you can see that grouping weekly with a threshold of 3 produces false positives.

[image]

Presumably, we're going to want this kind of summary for every data type we produce and ask for recommendations about best settings.

bpbond, 2025-05-22:
[image]

bpbond, 2025-06-10:
>Things on my mind from working with the L1 data so far include converting groundwater pressure to water table level (or surface water level) and converting soil conductivity to salinity if feasible

Can someone point me to the conversion step(s) for these?

nickdward, 2025-06-10:
> > Things on my mind from working with the L1 data so far include converting groundwater pressure to water table level (or surface water level) and converting soil conductivity to salinity if feasible
> 
> Can someone point me to the conversion step(s) for these?

I vote for @peter.regier@pnnl.gov for the definitive groundwater elevation calculations in R. I can help for soil salinity if needed, though my calculations are excel based currently. It’s a fairly simple temperature dependent calculation with a handful of coefficients.

Remind us when we’re back from deluge

bsulman, 2025-06-11:
I'm working on some python code for converting conductivity to salinity based on a doc that @Fausto2504 sent me. 

[Converting Terors EC to Pore water salinity.docx](https://github.com/user-attachments/files/20690532/Converting.Terors.EC.to.Pore.water.salinity.docx)

It sets up a linear interpolation for finding the roots of this polynomial equation for VWC because it was a lot slower solving for that every time. There may be an easier way to do it in R.

```
VWCs=np.linspace(0,1,100)
vwcs=np.array([np.polynomial.Polynomial((-10.848-VWC,1.302e-2,-5.105e-6,6.771e-10)) for VWC in VWCs])
raws=np.array([vwc.roots()[np.isreal(vwc.roots())].real for vwc in vwcs])
def cond_to_salinity_teros(args):
    (cond,T,VWC)=args
    if not (np.isfinite(cond) or np.isfinite(T) or np.isfinite(VWC)):
         return np.nan
    # Convert conductivity to porewater conductivity
    ep=80.3-0.37*(T-20)
    esb0=4.1
    # vwc=np.polynomial.Polynomial((-10.848-VWC,1.302e-2,-5.105e-6,6.771e-10))
    # raw=vwc.roots()[np.isreal(vwc.roots())].real
    raw=np.interp(VWC,VWCs,raws.squeeze())
    eb=(2.887e-9*raw**3 - 2.080e-5*raw**2 + 5.276e-2*raw - 43.39)**2
    cond_porewater=(ep*cond)/(eb-esb0)
    salinity=-2.060e-1 + 5.781e-4*cond_porewater
    return salinity
```

bpbond, 2025-06-11:
Thanks @bsulman !  I'll circle back with you guys -- including @nickdward and @peterregier -- next week.

bpbond, 2025-06-21:
Thanks @bsulman and @Fausto2504 -- I now have salinity as a derived variable in the Level 2 data following the steps above. 🥳 

@peterregier -- can you tell me about "the definitive groundwater elevation calculations in R"? If easier to discuss on Teams, I can put a meeting on our calendars this coming week.