COMPASS-DOE/sensor-data-pipeline #281: SWH_UP (now Swamp) sap flow deep sensor install issue

aestearns01, 2025-03-23:
@bpbond @stephpenn1 @roylrich  I noticed during intensive site checks on 18 Mar 25 that the SWH_UP (now swamp) SF 2 deep sensor was installed incorrectly with the heated probe on the bottom. Not sure how this will affect the data. SF 2  shallow 3.5 cm sensor was installed correctly. 

Additionally, how have we addressed the UP to Swamp naming convention in the L1 data set since in loggernet it is technically still UP? For monitoring, I still use "UP" to reflect the loggernet designations. Reminder that we don't have SF sensors in the current Upland (previously Upland Control).

bpbond, 2025-03-24:
Hi @aestearns01 

Oh, that's a thing. Presumably then the reference probe is seeing heated sap? Ugh.

Re your question, we use a 'design table' to map loggernet variables to L1 outputs, so we can simply assign different mappings:

[image]

Here the first block (`WaterLevel600A`) is mapped to SWAMP, while `WaterLevel600B` is assigned to UP.

aestearns01, 2025-03-24:
@bpbond yes, the ref probe is logging heated sap

Thanks for showing me how it looks!

stephpenn1, 2025-03-25:
Interestingly sensor 02D looks pretty good compared to the other deep sensors

[image]

aestearns01, 2025-03-25:
@stephpenn1 interesting. I suppose that's why I never caught it doing checks. I would have expected the daily flux to be opposite, no??

bpbond, 2025-03-26:
That _is_ interesting! Thanks @stephpenn1 

@aestearns01 No, I don't think it would be inverted or anything. The unheated probe is serving as a reference, so if it's slightly warmer than it should be (because it's above, not below, the heated one) all that would happen is that the computed sap flow would be a bit lower than the true value.

stephpenn1, 2025-03-26:
@aestearns01, if you're planning to re-install in the same tree, we could compare the two data sets. I'm curious how _much_ dampening is happening

bpbond, 2025-04-08:
@stephpenn1 

What about if we add a note to the SWH site metadata file, noting this problem but also that visual inspection of the data shows that it's performing like the others?

bpbond, 2025-04-09:
Please look at https://github.com/COMPASS-DOE/sensor-data-pipeline/pull/296/files -- does this suffice?

stephpenn1, 2025-04-09:
Yes that's good for now

closed 2025-04-09 by bpbond

aestearns01, 2025-04-11:
@bpbond it's the 02D (deep 5.0 cm sensor) that is installed incorrectly. The metadata note you created says the 02 sensor (regular 3.2 cm)
@stephpenn1 not planning to re-install any deep sensors

bpbond, 2025-04-11:
Ahhhhh! Thank you @aestearns01 -- much appreciated-- fixed in b3649a9273ae0567603ae68d662d095ba23e0822

PR #296 (2025-04-09): Add metadata note about mis-installed 02 sapflow sensor
Closes #281