COMPASS-DOE/sensor-data-pipeline #286: Exo sonde (open water) issues

Fausto2504, 2025-03-28:
- [ ] (1) Swapping data columns in LE sites (CRC, OWC, PTR): the Campbell loggers have swapping data between variables (e.g., pH=300 referred to ORP and not pH), which is less noticeable when the magnitude is similar (e.g., DO in mg/L vs. pH). This issue affects open water levels and multiple open water quality parameters. I am currently working on flagging them.

- [ ] (2) We should consider removing the Exo depth from L1. There is no Flag_out_of_water yet. CRC dries out on a daily or weekly basis. PTR 2022 and MSM were deployed in a buoy; PTR 2023 has no reference. The reference for OWC needs to be revised. I am planning to meet with folks to talk about it. 

- [ ] (3) Conductivity unit mismatch. Were values transmitted in mS/cm and not in µS/cm as expressed in the metadata? Expected conductivity data would be ~ 500 µS/cm in LE sites, but values are ~0.5. The low resolution (in mS/cm) is rounding numbers and reporting similar values on an hourly basis for conductivity-related parameters (e.g., salinity, etc.). @stephpenn1 @roylrich

bpbond, 2025-03-30:
Re the first point above, things certainly look bizarre:

[image]

bpbond, 2025-04-03:
Broke these into separate issues for ease of discussion and triage

closed 2025-04-03 by bpbond