COMPASS-DOE/sensor-data-pipeline #287: PTR ClimaVue PAR tot wrong 2022 to Mar 22, 2023

Fausto2504, 2025-03-28:
- [ ] PTR site show wx_par_tot24 (mmol/m2) values ~4 times higher from 2022-03-17 to 2023-03-22. The problem was fixed in 2023-03-22.

bpbond, 2025-03-30:
@Fausto2504 Here's the `wx_par_tot24` data from the three Lake Erie sites, from L1 v1-2:

[image]

I don't understand (or see) what you're flagging -- can you elaborate?

Fausto2504, 2025-03-31:
Sorry, I included a wrong variable name. 
PTR wx_par_tot15 show data much higher from "2022-03-15 15:00:00" to "2023-03-08 00:00:00" and no data from "2023-03-08 00:00:00" to "2023-03-23 10:00:00"

Fausto2504, 2025-03-31:
[image]

bpbond, 2025-03-31:
Thank you. A broader look at this:

[image]

So it looks like CRC and OWC had the same problem, presumably in their datalogger programs, but it was corrected earlier; the PTR site didn't get fixed until March 2023.

Is this a known issue @roylrich ?

bpbond, 2025-04-01:
Assigning to @roylrich for feedback on fixing old files by dividing by four.

roylrich, 2025-04-01:
This has something to do with a sampling rate change or not dividing by correct increment. This was not a known issue.  Ill check code and then look to see what can be done. 
Sorry and thanks. 
RR

bpbond, 2025-04-01:
No worries, and thank you! As you check, here are my two questions:
* Are all dataloggers now running fixed code? It looks like it based on outputs.
* For correcting past data flagged by @Fausto2504 above, will a simple division will do the trick?

bpbond, 2025-04-01:
@stephpenn1 I have assigned this to `L1 v2-0` to take advantage of [#621 ](https://github.com/COMPASS-DOE/sensor-data-pipeline/pull/261)

roylrich, 2025-04-02:
All, I hate Par_Totalized because it is subject to both the scan rate (how often scans run) and the totalized for the table period. 
Righ now we are doing the following the Climavue table at a 15 minute reporting interval:. So we are totalizing Par_total for 15 minutes based a 1 minute cycle. My hunch is that early programs ran at a 15 sec rate... and thus have a high totalization (i will check this)

 Average(1, PAR_Den_C, FP2, 0)
 Totalize(1, PAR_Tot_C, FP2, 0) 

Code running at SWH now. 

VoltDiff(PAR_Den_C,1,mV200,1,True,0,_60Hz,1,0) 'Use 20 mV range for CR3000, CR5000
      'Set negative values to zero
      If PAR_Den_C<0 Then PAR_Den_C=0
      PAR_Tot_C= PAR_Den_C* (Par2C /1000) *60 ' 60 (1 minute rate) is scan rate and needs to be changed with program changes
      PAR_Den_C= PAR_Den_C* Par2C 'Multipliers will differ for each sensor and scan rate.

Most people are only interested in daily totalized PAR so I think it would be better which is something to think about. We can rework these data from Par_den_c if needed.

roylrich, 2025-04-02:
Did you all get more information from me on this. I thought I put together a reply with code on Par_Tot and I am not seeing it in chain.

bpbond, 2025-04-02:
Taking a closer look at CRC-W. The change happens in late June 2022:

[image]

at the same time the number of observations per day quadruples:

[image]

bpbond, 2025-04-02:
I _think_ this is pretty conclusive -- for a while there were 24 "15 minute" observations per day, each four times as long as they should have been, and then it changed to 96 (one per 15 minutes). I'm going to prepare a PR fixing this.

PR #288 (2025-04-09): Correct wx_par_tot15 in old data
This PR corrects old values of `wx_par_tot15` that were being incorrectly calculated by the datalogger code, as flagged by @Fausto2504 in #287 .

Values are corrected up to following timestamps:
| Site(s) | Latest correction |
|--------|--------|
| CRC, OWC | 2022-06-28 10:30:00 |
| PTR | 2023-03-10 10:30:00 |
| GWI | 2024-08-23 00:00:00 |
| MSM | 2023-01-12 00:00:00 | 

Resulting time series (compare with figures in #287):

[image]

[image]

Closes #287
closed 2025-04-09 by bpbond