COMPASS-DOE/sensor-data-pipeline #290: Columns incorrectly swapped in some sonde data

bpbond, 2025-04-03:
Originally in #286 @Fausto2504 said

>(1) Swapping data columns in LE sites (CRC, OWC, PTR): the Campbell loggers have swapping data between variables (e.g., pH=300 referred to ORP and not pH), which is less noticeable when the magnitude is similar (e.g., DO in mg/L vs. pH). This issue affects open water levels and multiple open water quality parameters. I am currently working on flagging them.

bpbond, 2025-04-10:
This isn't showing the characteristics of swapped columns, at least not obviously. Here's `sonde_pH` for the LE sites:

[image]

Zooming in:

[image]

bpbond, 2025-04-10:
Right now, I'm not seeing a column swap, which would be characterized by _consistent_ nonsensical values, but rather a sensor that seems to be giving good data most (98+%) of the time...but occasionally not.

Anyway @Fausto2504 I would appreciate your thoughts and/or further investigation. Thanks!

Fausto2504, 2025-04-10:
@bpbond, the pH range between 5 and 10 works fine. But I have do a lot of QAQC, so I need to send you a document.

bpbond, 2025-04-10:
OK! If you email please cc @stephpenn1 so she stays in the loop.

Fausto2504, 2025-04-10:
Here are the flags I use:

1)	flag_out_range_
Excludes impossible data values based on observations of data ranges from each location.

df_exo <- df_exo %>%
  mutate(
    flag_out_range_depth = ifelse(depth_m < -5, TRUE, FALSE),
    flag_out_range_ph = ifelse(ph > 10 | ph < 5, TRUE, FALSE),
    flag_out_range_temp = ifelse(temp_c > 50, TRUE, FALSE),
    flag_out_range_orp = ifelse(orp_mv > 1000 | orp_mv < -1000 , TRUE, FALSE),
    flag_out_range_fdom_qsu = ifelse(fdom_qsu > 150, TRUE, FALSE),
    flag_out_range_fdom_rfu = ifelse(fdom_rfu > 100, TRUE, FALSE),
    flag_out_range_cond = ifelse(site=="MSM" & nlf_cond_micro_s_cm > 40000 |
                                 site=="MSM" & nlf_cond_micro_s_cm < 5000 |
                                 site!="MSM" & nlf_cond_micro_s_cm > 2000 |
                                 site!="MSM" & nlf_cond_micro_s_cm < 170, TRUE, FALSE)
  )

2)	flag_wrong_outliers 
Exclude data that was exchanged by the data loggers. It is based on the observations that when it happen there is at least one NA in one of the variables.

flag_wrong_outliers = ifelse(
is.na(odo_mg_l) | is.na(p_h) | is.na(p_h_mv) |
is.na(conductivity) | is.na(nlf_conductivity) | is.na(specific_conductivity_u_scm) |
is.na(fdom_qsu) | is.na(fdom_rfu) | is.na(odo_sat) | is.na(odo_local) |
is.na(pressure_psia) | is.na(salinity_ppt) | is.na(wiper_position_mv) |
is.na(temp_c) | is.na(depth_m),
TRUE, FALSE
)

3)	flag_out_of_water 
Exclude dates we know Exo was out of water. Note I don't have a record of MSM, but LE records are being recorded here:
https://docs.google.com/spreadsheets/d/1y08XAOzzpLZbIOMJquVUv-jSHXhHJpF9neG5kcJpnjQ/edit?gid=913708372#gid=913708372
# flag out of water dates
df_exo <- df_exo %>%
    mutate(flag_out_of_water = case_when(
    # CRC - 2022
    site == "CRC" & year == 2022 & (date < as.Date("2022-07-21") | date > as.Date("2022-10-20") | date == as.Date("2022-07-19")) ~ TRUE,
    # CRC - 2023
    site == "CRC" & year == 2023 & (date < as.Date("2023-04-15") | date > as.Date("2023-11-16") | date == as.Date("2023-07-10") | date == as.Date("2023-09-01")) ~ TRUE,
    # CRC - 2024
    site == "CRC" & year == 2024 & (date < as.Date("2024-04-25") | date > as.Date("2024-11-05")) ~ TRUE,
    # PTR - 2022
    site == "PTR" & year == 2022 & (date < as.Date("2022-08-03") | date > as.Date("2022-10-19")) ~ TRUE,
    # PTR - 2023
    site == "PTR" & year == 2023 & (date < as.Date("2023-06-24") | date > as.Date("2023-11-24")) ~ TRUE,
    # OWC - 2022
    site == "OWC" & year == 2022 & (date < as.Date("2022-06-22") | date > as.Date("2022-11-11") | date == as.Date("2022-10-05")) ~ TRUE,
    # OWC - 2023
    site == "OWC" & year == 2023 & (date < as.Date("2023-04-12") | date > as.Date("2023-11-16")) ~ TRUE,
    # OWC - 2024
    site == "OWC" & year == 2024 & (date < as.Date("2024-03-30") | date > as.Date("2024-12-06") | date == as.Date("2024-04-05")) ~ TRUE,
    # Default case if none of the conditions are met
    TRUE ~ FALSE
  ))

4)	flag_out_of_SPC
Exclude data from when the specific conductivity (nlf_cond_micro_s_cm) vary in a range that indicates that the water level was below the water quality sensors.  
Notes: Conductivity values need to be in micro S/cm, which is not in L1 data.
This helps to get good data for water quality parameter. But, as pressure is above the conductivity sensor it does not fully exclude data when pressure sensor was out of water.
CRC and PTR in late fall the probes are out of water very often. 
# flag out of water based on conductivity variability
df_exo <- df_exo %>%
  filter(flag_out_of_water == FALSE,
         flag_out_range_cond == FALSE
         ) %>%
  mutate(
    diff_previous_cond = abs(nlf_cond_micro_s_cm - lag(nlf_cond_micro_s_cm)),
    diff_next_cond = abs(nlf_cond_micro_s_cm - lead(nlf_cond_micro_s_cm)),
    diff_previous_cond = replace_na(diff_previous_cond, 0),
    diff_next_cond = replace_na(diff_next_cond, 0),
    flag_out_of_SPC = case_when(
      site != "MSM" & (diff_previous_cond > 20 | diff_next_cond > 20) ~ TRUE,
      site == "MSM" & (diff_previous_cond > 1000 | diff_next_cond > 1000) ~ TRUE,
      TRUE ~ FALSE
    )
  )  )

Fausto2504, 2025-04-11:
2)	flag_wrong_outliers 
Exclude data that was exchanged by the data loggers. It is based on the observations that when it happen there is at least one NA in one of the variables.

flag_wrong_outliers = ifelse(
is.na(odo_mg_l) | is.na(p_h) | is.na(p_h_mv) |
is.na(conductivity) | is.na(nlf_conductivity) | is.na(specific_conductivity_u_scm) |
is.na(fdom_qsu) | is.na(fdom_rfu) | is.na(odo_sat) | is.na(odo_local) |
is.na(pressure_psia) | is.na(salinity_ppt) | is.na(wiper_position_mv) |
is.na(temp_c) | is.na(depth_m),
TRUE, FALSE
)

bpbond, 2025-04-11:
Brief coding aside, not about science. You don't need to write
```r
results <- ifelse(x | y | z, TRUE, FALSE)
```
Just do
```r
results <- x | y | z
```

Fausto2504, 2025-11-18:
Since most of cases the file generates a NAN in one of the columns, for my manuscript I use this flag:

flag_wrong_outliers = ifelse(
is.na(odo_mg_l) | is.na(p_h) | is.na(p_h_mv) |
is.na(conductivity) | is.na(nlf_conductivity) | is.na(specific_conductivity_u_scm) |
is.na(fdom_qsu) | is.na(fdom_rfu) | is.na(odo_sat) | is.na(odo_local) |
is.na(pressure_psia) | is.na(salinity_ppt) | is.na(wiper_position_mv) |
is.na(temp_c) | is.na(depth_m),
TRUE, FALSE
)