COMPASS-DOE/sensor-data-pipeline #292: Eco conductivity unit mismatch?

bpbond, 2025-04-03:
Originally in #286 @Fausto2504 said

>(3) Conductivity unit mismatch. Were values transmitted in mS/cm and not in µS/cm as expressed in the metadata? Expected conductivity data would be ~ 500 µS/cm in LE sites, but values are ~0.5. The low resolution (in mS/cm) is rounding numbers and reporting similar values on an hourly basis for conductivity-related parameters (e.g., salinity, etc.). @stephpenn1 @roylrich

bpbond, 2025-04-11:
87% of the `gw_act_cond` data for the LE sites fall within a 500-6000 range:

[image]

Most of the outliers consist of very high (a cluster around 50 000, and another at 100 000) values at PTR.

However, 6.8% of the data are very low:

[image]

...spread across all the trolls (except for PTR TR, where it almost never happens):

  Site  Plot  Instrument_ID `n()`
  <chr> <chr> <chr>         <int>
1 CRC   TR    B              6834
2 CRC   UP    A              8834
3 CRC   W     C              8467
4 PTR   TR    B                33
5 PTR   W     C              4587

bpbond, 2025-04-11:
In summary, I don't see any evidence of a units problem, as ~90% of the LE sites' conductivity data fall within the expected (?) range, although there clearly are a small number of goofy (very low or very high) values.

But I'm not an expert here and welcome further thoughts @Fausto2504 -- thanks!

Fausto2504, 2025-04-11:
Sorry. I did not specified that I was indicating: 

sonde_spcond         µS/cm       µS/cm     0, 1e+05     Water specific conductivity

bpbond, 2025-04-11:
Thanks for clarifying.

[image]

For `sonde_spcond` about 1.5% of the data are very high, but most (97.8%) fall within a small range:

[image]

The data system isn't applying any units transformation here--these are the numbers coming off the datalogger:

[image]

Fausto2504, 2025-04-11:
MSM conductivity is ~20-40 times higher than in LE sites because of salinity. 

MSM expected values are ~20,000 uS/cm or ~ 20 mS/cm. 

LE expected values are ~500-1000 uS/cm or ~ 0.5-1 mS/cm.

Data logger's configuration is wrongly set for mS/cm and L1 report as microS/cm. 

Note: The solution could be easily multiplying by 1000; however, this resolution make LE specific conductivity with "rectilinear variability" in hourly variation.

Fausto2504, 2025-04-11:
The solution could be easily multiplying by 1000; however, this resolution make LE specific conductivity with "rectilinear variability" in hourly variation.

Fausto2504, 2025-04-11:
Ranges for specific conductivity that I use that work. But because specific conductivity is in a bad unity. I am using nlf_conductivity, but ranges are very similar (note code is still with the if else true and false you suggested to I change:

5,000 microS/cm (or < 5 mS/cm) < MSM < 40,000 microS/cm (or < 40 mS/cm)
100 microS/cm (or < 0.1 mS/cm) < CRC, PTR, OWC < 2,000 microS/cm (or < 2 mS/cm)

flag_out_range_cond = ifelse(site=="MSM" & nlf_cond_micro_s_cm > 40000 |
                                 site=="MSM" & nlf_cond_micro_s_cm < 5000 |
                                 site!="MSM" & nlf_cond_micro_s_cm > 2000 |
                                 site!="MSM" & nlf_cond_micro_s_cm < 100, TRUE, FALSE)

bpbond, 2025-11-12:
That's my understanding. @Fausto2504 ?

Fausto2504, 2025-11-18:
Double-checked and **all Exo Conductivities are in mS/cm**: Conductivity, Specific Conductivity and NLF Cnductivity (for all sites).