COMPASS-DOE/sensor-data-pipeline #293: Sensor inventories with L1 releases?

stephpenn1, 2025-04-03:
Mapping L1 sensor data to location metadata (i.e. sapflow tree inventory) is currently confusing and there are multiple versions on github/g drive.

Maybe we should produce a csv from the monitoring sheets for each release to ensure accurate sensor information. Because for example, let's say a sap flow sensor dies after an L1 release. If someone joined the sap flow monitoring sheet to that L1 release, it wouldn't have information for the now-dead sensor.. does that make sense?

bpbond, 2025-04-08:
That does make sense. 
1. These files would _not_ appear in the main L1 data files, but rather be available as supplementary information (probably in their own folder), right?
2. We're basically talking about the various monitoring spreadsheets that provide detailed information about inventory tag numbers, etc.?

stephpenn1, 2025-04-08:
Correct,

And they could even just be included somewhere internally or on ess-dive (with L1 package or in a separate package?). But either way making versioned inventory sheets that correspond to the L1 releases could be useful

bpbond, 2025-04-08:
OK, what kind of information would these have?

| Data type | Supplementary info |
|--------|--------|
| Sapflow | Tree tag number, date installed? |
| Aquatroll | Date installed? |
| Teros | Date installed? |
| Exo | Date installed? |
| Sonde | Date installed? |

robertapeixoto, 2025-04-09:
DBH_Year